eclipse-embed-cdt/eclipse-plugins #85: CMSIS-Pack manager: Kinetis KLxx Devices do not show up even if installed

ErichStyger, 2016-02-20:
Using:
- Window 7, 64bit,
- Kinetis Design Studio V3.1 (Eclipse Luna)
- GNU ARM C/C++ Packs (Experimental)	2.2.2.201602101653 ilg.gnuarmeclipse.packs.feature.group

I have the 'Kinetis_KLxx_DFP' pack installed from http://www.keil.com/pack/index.idx (see screenshots)
[image]

As well under 'Devices' the 'KLxx Series' is grayout out, but installed:

Exit and restart Eclipse did not help.
[image]

Let me know if you need any further information.

Thanks,
Erich

ilg-ul, 2017-04-25:
this is a particular case of #203

ErichStyger, 2022-01-25:
cleaning up old issues, I think this one could be closed.

closed 2022-01-25 by ErichStyger