eclipse-embed-cdt/eclipse-plugins #88: Use multiple line for excluded file

dumarjo, 2016-02-25:
Hi,

I would like to know if it's possible to modify the .cproject to have the excluded files section to be on multiple line instead of 1 single line. We have problem  with that line with versionning control like SVN or TFS. 

do you think that this is possible ?

Regards

Jonathan

ilg-ul, 2016-02-25:
the excluded files section is managed by CDT, not by the GNU ARM Eclipse plug-ins.

diabolo38, 2016-02-25:
what eclipse and svn version do you used and what type of issue are you getting? 
i have quite a few project on svn  with several file "excluded from build"  and didn't get any issue ?
As far i can remind  it was ok for on Kepler and up to luna . I'm not sure i have done any check in/out since i moved to mars.

dumarjo, 2016-02-25:
The problem is when 2 people add/remove folder from excluded from build, we always have a conflit on the excluded ressources line in the .cproject. 

We have eclipse kepler. The project was build on indigo i think. This is a 2 years old project.

regards

Jonathan

diabolo38, 2016-02-25:
I see, it's quite normal so, and It should be pretty simple to merge the conflicting changes . 
but maybe not in eclipse :(

dumarjo, 2016-02-26:
Hi,

When a project have a small amount of compile target and a small amount of exclude build i's easy to merge. But now we have a project with 5 compile target and more than 150 cpp file in many folder. 

I Have do a request on the bugzilla of CDT for that. 

Regards

Jonathan

closed 2017-04-24 by ilg-ul