eclipse-embed-cdt/eclipse-plugins #89: Pack manager: Nordic devices do not show up in project preferences

julesvo, 2016-03-08:
Hi,

After succesfully installing the Nordic Semiconductor nRF_DeviceFamily pack through the Packs plugin (version 2.2.2.201602101653) in Eclipse 4.5.2, I'm not able to see the devices in my project settings. I've tried different versions of the nRF_DeviceFamily pack in combination with or without the other Nordic packs. I tried different projects and restarting Eclipse and my pc, however, I do not seem to get this working. Can anyone help me?

Thanks,

Jules

ilg-ul, 2016-03-08:
which pack do you suggest to try and which devices are you expect to see, so I can reproduce the problem?

julesvo, 2016-03-08:
Hi,

I am using the Nordic NRF52 DK, PCA10040 with a nRF52832 chip. I installed the newest available DeviceFamily pack: 8.5.0.

Thanks for helping me,

Jules

ilg-ul, 2016-03-08:
I installed the Nordic DFP (which, btw, is not 8.5.0), and the nRF52832 showed under Devices.

What else do you expect to happen?

Currently the device description can be used only during debug, to show the peripheral registers, if you assign devices to projects.

julesvo, 2016-03-08:
Hi, what version of DFP do you use? If I use 8.5.0 (the newest in my list?) I do not see any devices appearing under my project settings, but I think I found a fix to solve it. If I go to the .cache directory in my Packages directory, I see the "NordicSemiconductor.nRF_DeviceFamilyPack.8.5.0.pack ", but the NordicSemiconductor.nRF_DeviceFamilyPack.8.5.0.pdsc file is missing. I manually added this file by opening the NordicSemiconductor.nRF_DeviceFamilyPack.8.5.0.pack with Winrar and copying this file to the .cache dir. Now I can see the device and see the peripherals when debugging. I think this should be done automatically but maybe I'm wrong.

ilg-ul, 2016-03-08:
> what version of DFP do you use?

8.3.2 is the latest in my list

it looks like an out of sync problem, the Keil index is most of the time not synced with the Nordic server, and the plug-in does not cope very well with this. :-(

vidarbe, 2016-03-10:
Sorry, I overlooked this report somehow. Filed a duplicate on this, see issue #92. To get the index in sync I have used to delete the nordic files from the packs folder before doing a refresh.

Blackclaws, 2016-10-28:
I can confirm that I have the same problem and that the fix described in issue #92 does work. It appears to be either an error of how the packs are parsed or extracted or a change in how Nordic creates those packages.

ilg-ul, 2016-10-28:
as I already said, the Keil index is most of the time not synced with the Nordic server.

the next version of the Packs Manager will try to be more resilient to these problems, but it'll take som e time to be ready.

Blackclaws, 2016-10-28:
Ah interesting. I didn't think that the Keil Index would refer to the contents of the packs itself just to the download locations. I thought the packs itself would contain all the needed information to use them.

ilg-ul, 2016-10-28:
the index also contains a version; I make the mistake to trust it :-(

Blackclaws, 2016-10-28:
Well aside from that little problem everything works as expected afterwards. Keep up the good work. I really like the plugin and use it a lot.

closed 2017-04-24 by ilg-ul