eclipse-embed-cdt/eclipse-plugins #95: CMSIS Pack Manager: would be nice to import packages from local repository

marek-trmac, 2016-03-15:
e.g. support URL in with "file:" scheme.
User can easily create repository on his local machine and import selected packages.

ilg-ul, 2016-03-15:
did you try it? the code to process the url is generic, I thought it works with any url.

you need to provide an index file, similar to the one from keil.com.

marek-trmac, 2016-03-15:
The problem is in ilg.gnuarmeclipse.packs.data.Repos.getFileNamePrefixFromUrl(String)
1) ':' character is not converted
2) file:/// prefix is not removed
3) '\' character is not supported, cannot be used in URL

in4lio, 2016-03-15:
Hello, I confirm local repo does't work.

ilg-ul, 2016-03-15:
> 3) '\' character is not supported, cannot be used in URL

that's normal, to avoid problems, in Eclipse you should always use POSIX paths, even on Windows.

in4lio, 2016-03-15:
But, for now, it's possible to add the package into the.cache and manually update .repos.xml)

marek-trmac, 2016-03-15:
Fixed by pull request #95

ilg-ul, 2016-03-15:
if you confirm the version from updates-test is ok, we can close this ticket.

marek-trmac, 2016-03-24:
updates-test works well.
For a reference, here is example how URL should look like:
`file:///c:/nxp/CMSIS_Pack_Repo/index.idx`

closed 2016-03-24 by marek-trmac