eclipse-embed-cdt/eclipse-plugins #100: cannot find -lstdc++_nano, -lg_nano, -lc_nano

sledgeh, 2016-03-16:
Cannot compile example project. System Manjaro Linux. Toolchain was installed from [AUR](https://aur.archlinux.org/packages/arm-none-eabi-gcc49-linaro/)

```
Building target: blinker-f4.elf
...
/usr/lib/gcc/arm-none-eabi/4.9.2/../../../../arm-none-eabi/bin/ld: cannot find -lstdc++_nano
/usr/lib/gcc/arm-none-eabi/4.9.2/../../../../arm-none-eabi/bin/ld: cannot find -lg_nano
/usr/lib/gcc/arm-none-eabi/4.9.2/../../../../arm-none-eabi/bin/ld: cannot find -lc_nano
collect2: error: ld returned 1 exit status
```
Path where it wants to find libs doesn't exist.
I tried to fix it in `Project Settings` -> `C/C++ Build` -> `Settings` -> `Tool Settings` -> `Cross ARM C++ Linker` -> `Libraries`. I've added pathes where stdlib were found `/usr/arm-none-eabi/lib/`. There is no nano libs at all. But that path have `libstdc++.a`.
How to set non-nano stdlib?

sledgeh, 2016-03-16:
Fixed by disabling flag `use newlib-nano (--specs=nano.specs)`
So newlib nano does not exist because toolchain builder has skip it. Please help if you know how to fix this non-bug?

closed 2016-03-16 by sledgeh

ilg-ul, 2016-03-16:
of course you are free to experiment with any toolchain you want, but support is available only for the officially supported toolchains, available from launchpad/linaro.

if you want to save you some time, please use only supported toolchains.

sledgeh, 2016-03-16:
This is Linaro toolchain.

ilg-ul, 2016-03-16:
did you try to download the latest archive from the linaro site? (http://www.linaro.org/downloads/)

if so, does it behave the same?

on the other hand, the nano version might be available only for bare metal builds, you must read the documentation that accompany your toolchain.

sledgeh, 2016-03-16:
Understand. I will compile the lastest toolchain soon. Now it works.