eclipse-embed-cdt/eclipse-plugins #101: Invalid arguments ' Candidates are: void sleep(?) '

sledgeh, 2016-03-16:
I all examples in `main.cpp` file in all lines where `timer.sleep` present Eclipse warns with message: Semantic error *Invalid arguments ' Candidates are: void sleep(?) '* and shows pictogram "bug". Toolchain can compile it. 
How can I shut up Eclipse?

sledgeh, 2016-03-16:
I fixed it while composing message.
Disable Preferences -> C/C++ -> Code Analysis -> Syntax and Semantec Errors -> Invalid arguments or all group. 
May be this is real semantec error and should be fixed? Or Eclipse is not enough clever?

closed 2016-03-16 by sledgeh

ilg-ul, 2016-03-17:
did you try project properties -> Index -> Rebuild ?

sledgeh, 2016-03-17:
Many times.

ilg-ul, 2016-03-28:
from my experience on OS X, these errors appear only in the initial project, after reindexing and closing/opening the files, they are no longer shown.

sledgeh, 2016-03-30:
Yes, they appear only in the initial project. And reindexing fixes the problem. On Linux too.

daveloyall, 2017-02-01:
And windows.  Rebuilding the index does fix it.

caot, 2018-06-21:
`project properties -> Index -> Rebuild` also fixed the follows:
`Problem description: Invalid arguments ' Candidates 
 are: void __builtin_va_end(char * (*)()) '`

positron96, 2019-12-04:
Sorry for digging this up, but where is Index -> Rebuild in the current version of Eclipse?

[image]