eclipse-embed-cdt/eclipse-plugins #102: pyOCD launch config panel doesn't support gdbserver being on PATH

sukidog, 2016-03-17:
For users who add pyOCD to PATH, they should be able to specify the gdbserver executable as a simple filename (sans path). Currently this isn't supported.

I'll be contributing the fix shortly.

ilg-ul, 2016-03-17:
GNU ARM Eclipse plug-ins discourage the use of system path for various tools.

for PyOCD, a for the other debugging plug-ins, you have to set the path in the specific Preferences page, below Run/Debug.

sukidog, 2016-03-17:
How PATH Is set is orthogonal to this issue

John

sukidog, 2016-03-17:
Actually, disregard my last comment. I think you are saying you don't want the tools to cater or even support programs being in PATH

John

ilg-ul, 2016-03-17:
then what is the problem? 

if you do not want to use the path macro, remove it form the field. same for all debugging plug-ins.

ilg-ul, 2016-03-17:
> you don't want the tools to cater or even support programs being in PATH

no, I want to say that adding a program to your user shell path does not guarantee to be found in a gui application. I don't know on windows, but for some unix boxes (my mac and several linux distros I tried) you need to set a separate path for gui applications (a separate file).

to avoid such odd behaviours (and support tickets), my plug-ins require an explicit path to each tool.

sukidog, 2016-03-17:
Well, I've never run into something like that. A GUI app ultimately just
uses an OS API to invoke the cmdline program. From my experience, that
program should work the same as if the program is invoked from a shell
(assuming both the shell and the GUI app have the same environment
variables). A specific problem example would help clear up my
confusion/doubt.

sukidog, 2016-03-18:
I guess the bottom line here is: will you reject a pull request that provides the behavior requested here? I.e., while you may want to discourage people from using PATH with Eclipse, will you outright prevent people from doing so by turning down logic that tries to support it (and probably works in all or a vast majority of cases...at least for pyOCD)?

ilg-ul, 2016-03-18:
> A GUI app ultimately just uses an OS API to invoke the cmdline program

yes, but in a different environment

> From my experience, that program should work the same as if the program is invoked from a shell (assuming both the shell and the GUI app have the same environment variables)

they don't. 

if you change the PATH in, let's say $HOME/.bashrc, this will be visible only for terminal shell sessions, the GUI may but usually may not use this file.

you need to change the PATH in some system wide location, specific to each platform.

> A specific problem example 

on OS X, setting a custom PATH for all graphical applications is possible by adding a PATH entry in $HOME/.MacOSX/environment.plist. by default this file is not present and needs to be created.

different linux distros may have totally different locations for this.

> will you reject a pull request that provides the behavior requested here? 

go ahead and prepare an Eclipse patch to change the system GUI behaviour on all platforms(!), and I'll consider it.

sukidog, 2016-03-18:
The things you are bringing up are challenges the *user* or the *tool
developer* has to contend with. I'm not saying your plugins should make
using PATH-based tools easy or reliable. I'm saying your plugins shouldn't
make it impossible. Right now, the pyOCD plugin makes it impossible. If the
string in the gdbserver field doesn't resolve to an absolute file path that
exists, then the code doesn't accept it. My fix is simple and has no
OS-specific code in it. It will work equally well for any OS. All it does
is give the user the *possibility* for things to work if he has gdbserver
in PATH and he specifies the exe by simple name. If it works for him,
great. If it doesn't, oh well--he'll need to specify a full path
(explicitly or via substitution variables). I'll create the pull request
and we'll see what you think

closed 2016-04-04 by ilg-ul