eclipse-embed-cdt/eclipse-plugins #108: J-Link Enable Flash Breakpoints checkbox doesn't change launch config

sukidog, 2016-03-30:
Fix on its way.

Steps to reproduce:
1. Create a J-Link launch configuraiton
2. Ensure the configuration is not dirty (the Apply button is disabled)
3. Switch to Startup tab and toggle 'Enable flash breakpoints'.
BUG: notice Revert and Apply buttons do not become enabled. You can close the launch config and reopen it and you'll see the setting did not change.

ilg-ul, 2016-04-04:
thank you, I'll investigate.

no need to post issues if you post pull requests.

sukidog, 2016-04-07:
Well, the fix is quite simple and in a pull request, so reviewing the fix may be a more efficient path than investigating the problem.

ilg-ul, 2016-04-07:
I'm quite busy now with other parts of the project, but I'll return to the plug-ins and apply your pull request.

ilg-ul, 2017-01-14:
fixed in v3.2.1-201701141320

closed 2017-01-14 by ilg-ul