eclipse-embed-cdt/eclipse-plugins #111: J-Link software breakpoints support

metc, 2016-04-06:
Hi,

I am not able to make working software breakpoints with J-Link probe. 

* The following code works with GDB and OpenOCD with a St-Link V2 probe (SWD).
```c
void __attribute__((noreturn)) __assert_func(const char *file, int line, const char *func, const char *failedexpr) {
  asm volatile("bkpt");
  abort();
}
```
* The following code works with a Segger J-Link Lite probe and Keil uVision 5:
```c
__asm void break_here(void)
{
    bkpt #0x01
    mov pc, lr  
}
```

* Unfortunately, the same code does not work with the GNU ARM C/C++ J-Link Debugging plugin in Eclipse. I get the following error, but the processor does not stop:

```
Read 4 bytes @ address 0x00023544 (Data = 0x46F7BE01)
Starting target CPU...
Read 4 bytes @ address 0x00023544 (Data = 0x46F7BE01)
Starting target CPU...
```

I use the following software versions :
 * Eclipse Luna Service Release 2 (4.4.2)
 * GNU ARM C/C++ J-Link Debugging version: 3.2.1.201511061603
 * Latest available Segger version V5.12

I also tested the same project using Eclipse Mars 2 and GNU ARM Eclipse plug-ins v2.11.3.

Based on the tests I made, I think the problem comes from the plugin, or a wrong configuration of it. Are the software breakpoints supported with the J-Link Debugging plugin ?

Any help is welcome.

Sources:
 * http://forum.segger.com/index.php?page=Thread&postID=10650&highlight=#post10650
 * https://devzone.nordicsemi.com/question/74116/software-breakpoints-with-j-link-probe/

ilg-ul, 2016-04-06:
I use bkpt in my code without problems, with the j-link plug-in, so it generally works, but I did not test it in your specific conditions.

The plug-in starts the j-link gdb server with some default options, but does not prevent you to add your own specific options if you want to enable other functionality.

And I have serious doubts that the plug-in can prevent gdb to stop at a breakpoint, if the hardware encounters one.

The plug-in is just a wrapper, you can start the gdb server with the options of your choice and also start the gdb client manually, with the options of your choice, to test if the software breakpoints are functional or not.

ErichStyger, 2016-04-07:
I can confirm as well from my side that bkpt instructions work for me without problems using the GNU ARM Eclipse (with Segger) plugins, both for Cortex M4 and M0+:
...Breakpoint reached @ address 0x00000DB0
Reading all registers
Removing breakpoint @ address 0x00000DB0, Size = 2
Read 4 bytes @ address 0x00000DB0 (Data = 0xFFCAF000)
Reading 64 bytes @ address 0x20002FC0
Starting target CPU...
...Target halted (PC = 0x00000DA4)
Reading all registers
Read 4 bytes @ address 0x00000DA4 (Data = 0x46BDBE00)
Reading 64 bytes @ address 0x20002FC0
Read 4 bytes @ address 0x00000DB8 (Data = 0xB08AB580)

metc, 2016-04-07:
Thank you @ilg-ul and @ErichStyger 
@ErichStyger what is your debugger / startup configuration ?

I used these initialization commands:
```
monitor reset halt
monitor mww 0xe0042004 0x00000001
monitor mww 0xe0042008 0x00001800
```

ilg-ul, 2016-04-07:
why do you need a custom configuration?

ErichStyger, 2016-04-07:
I don't need any specific startup configuration.
In the 'Debugger' Tab I do have:
Other options: -s
GDB client commands: set mem inaccessible-by-default off
In the startup tab: no specifc commands set. I have the check box 'enable flash breakpoints' enabled.

My full log:
SEGGER J-Link GDB Server V5.10n Command Line Version

JLinkARM.dll V5.10n (DLL compiled Feb 19 2016 18:39:11)

-----GDB Server start settings-----
GDBInit file:                  none
GDB Server Listening port:     2331
SWO raw output listening port: 2332
Terminal I/O port:             2333
Accept remote connection:      localhost only
Generate logfile:              off
Verify download:               on
Init regs on start:            on
Silent mode:                   off
Single run mode:               on
Target connection timeout:     0 ms
------J-Link related settings------
J-Link Host interface:         USB
J-Link script:                 none
J-Link settings file:          none
------Target related settings------
Target device:                 MKL25Z128xxx4
Target interface:              SWD
Target interface speed:        1000kHz
Target endian:                 little

Connecting to J-Link...
J-Link is connected.
Firmware: J-Link OpenSDA compiled Apr 24 2014 14:43:37
Hardware: V1.00
S/N: 621000000
Checking target voltage...
Target voltage: 3.30 V
Listening on TCP/IP port 2331
Connecting to target...Connected to target
Waiting for GDB connection...Connected to 127.0.0.1
Reading all registers
Read 4 bytes @ address 0x00000000 (Data = 0x20003000)
Target interface speed set to 1000 kHz
Resetting target
Halting target CPU...
...Target halted (PC = 0x00001520)
R0 = 00000000, R1 = A5A5A5A5, R2 = 00000000, R3 = 00000001
R4 = A5A5A5A5, R5 = A5A5A5A5, R6 = A5A5A5A5, R7 = 200007C0
R8 = A5A5A5A5, R9 = A5A5A5A5, R10= A5A5A5A5, R11= A5A5A5A5
R12= A5A5A5A5, R13= 20003000, MSP= 20003000, PSP= 200007C0
R14(LR) = 000057B5, R15(PC) = 00001520
XPSR 61000000, APSR 60000000, EPSR 01000000, IPSR 00000000
CFBP 00000000, CONTROL 00, FAULTMASK 00, BASEPRI 00, PRIMASK 00
Reading all registers
Select auto target interface speed (1429 kHz)
Flash breakpoints enabled
Downloading 192 bytes @ address 0x00000000 - Verified OK
Downloading 16 bytes @ address 0x00000400 - Verified OK
Downloading 15880 bytes @ address 0x00000410 - Verified OK
Downloading 8 bytes @ address 0x00004218 - Verified OK
Downloading 4 bytes @ address 0x00004220 - Verified OK
Downloading 4 bytes @ address 0x00004224 - Verified OK
Downloading 116 bytes @ address 0x00004228 - Verified OK
Downloading 24 bytes @ address 0x0000429C - Verified OK
Writing register (PC = 0x00000f38)
Read 4 bytes @ address 0x00000F38 (Data = 0x1C204C0A)
Read 2 bytes @ address 0x00000DB0 (Data = 0xF000)
Read 2 bytes @ address 0x00000DB0 (Data = 0xF000)
Read 2 bytes @ address 0x000039FE (Data = 0x4B10)
Read 2 bytes @ address 0x000039FE (Data = 0x4B10)
Resetting target
Halting target CPU...
...Target halted (PC = 0x00000F38)
Read 2 bytes @ address 0x00000DB0 (Data = 0xF000)
Read 2 bytes @ address 0x00000DB0 (Data = 0xF000)
Read 2 bytes @ address 0x00000DB0 (Data = 0xF000)
R0 = 00000000, R1 = F000300C, R2 = 00004400, R3 = 00008800
R4 = 1FFFF350, R5 = FFFFFFFF, R6 = 00000000, R7 = 1FFFF165
R8 = A5A5A5A5, R9 = A5A5A5A5, R10= A5A5A5A5, R11= A5A5A5A5
R12= A5A5A5A5, R13= 20003000, MSP= 20003000, PSP= 200007C0
R14(LR) = 1FFFF15D, R15(PC) = 00000F38
XPSR 41000000, APSR 40000000, EPSR 01000000, IPSR 00000000
CFBP 00000000, CONTROL 00, FAULTMASK 00, BASEPRI 00, PRIMASK 00
Reading all registers
Read 4 bytes @ address 0x00000F38 (Data = 0x1C204C0A)
Setting breakpoint @ address 0x00000DB0, Size = 2, BPHandle = 0x0001
Starting target CPU...
...Breakpoint reached @ address 0x00000DB0
Reading all registers
Removing breakpoint @ address 0x00000DB0, Size = 2
Read 4 bytes @ address 0x00000DB0 (Data = 0xFFCAF000)
Reading 64 bytes @ address 0x20002FC0
Starting target CPU...
...Target halted (PC = 0x00000DA4)
Reading all registers
Read 4 bytes @ address 0x00000DA4 (Data = 0x46BDBE00)
Reading 64 bytes @ address 0x20002FC0
Read 4 bytes @ address 0x00000DB8 (Data = 0xB08AB580)

ilg-ul, 2016-04-07:
(sorry Erich, wrong question; to avoid such misunderstandings, in aviation communication, each phrase starts with the call sign of the desired party) ;-)

@metc, why do you need a custom configuration?

J-Link defaults are fine for most cases, only when you debug in RAM, or debug larger A cores, you need to set custom configurations, but in this case you need to know very well what you are doing.

metc, 2016-04-07:
The default configuration does not work mor fe. I have the same configuration as @ErichStyger and I use the latest Segger tools (JLinkARM.dll V5.12). Here is my log output:
```
SEGGER J-Link GDB Server V5.12 Command Line Version

JLinkARM.dll V5.12 (DLL compiled Mar 30 2016 17:28:31)

-----GDB Server start settings-----
GDBInit file:                  none
GDB Server Listening port:     2331
SWO raw output listening port: 2332
Terminal I/O port:             2333
Accept remote connection:      localhost only
Generate logfile:              off
Verify download:               on
Init regs on start:            on
Silent mode:                   off
Single run mode:               on
Target connection timeout:     0 ms
------J-Link related settings------
J-Link Host interface:         USB
J-Link script:                 none
J-Link settings file:          none
------Target related settings------
Target device:                 nRF51822_xxAA
Target interface:              SWD
Target interface speed:        auto
Target endian:                 little

Connecting to J-Link...
J-Link is connected.
Firmware: J-Link Lite-Cortex-M V8 compiled Aug 20 2015 17:57:19
Hardware: V8.00
S/N: 518005903
Feature(s): GDB
Checking target voltage...
Target voltage: 3.33 V
Listening on TCP/IP port 2331
Connecting to target...Connected to target
Waiting for GDB connection...Connected to 127.0.0.1
Reading all registers
Read 4 bytes @ address 0x00000000 (Data = 0x000007C0)
Read 2 bytes @ address 0x00000000 (Data = 0x07C0)
Target interface speed set to 1000 kHz
Resetting target
Halting target CPU...
...Target halted (PC = 0x000006D0)
R0 = FFFFFFFF, R1 = FFFFFFFF, R2 = FFFFFFFF, R3 = FFFFFFFF
R4 = FFFFFFFF, R5 = FFFFFFFF, R6 = FFFFFFFF, R7 = FFFFFFFF
R8 = FFFFFFFF, R9 = FFFFFFFF, R10= FFFFFFFF, R11= FFFFFFFF
R12= FFFFFFFF, R13= 000007C0, MSP= 000007C0, PSP= FFFFFFFC
R14(LR) = FFFFFFFF, R15(PC) = 000006D0
XPSR C1000000, APSR C0000000, EPSR 01000000, IPSR 00000000
CFBP 00000000, CONTROL 00, FAULTMASK 00, BASEPRI 00, PRIMASK 00
Reading all registers
Read 4 bytes @ address 0x000006D0 (Data = 0x4C174916)
Read 2 bytes @ address 0x000006D0 (Data = 0x4916)
Select auto target interface speed (2000 kHz)
Flash breakpoints enabled
Semi-hosting enabled (Handle on BKPT)
Semihosting I/O set to TELNET Client
SWO disabled succesfully.
SWO enabled succesfully.
```

The target does not halt on this code:

```asm
 __asm volatile(
    "bkpt #0x01;"
    "mov pc, lr;"
);
```

ilg-ul, 2016-04-07:
I use "bkpt", I guess this is equivalent to "bkpt #0"; never tried "bkpt #1".

metc, 2016-04-11:
Sorry, I cannot make it working.  `bkpt`, `bkpt #0` or `bkpt #1`. The output is:
```
Starting target CPU...
...Breakpoint reached @ address 0x000236A4
Reading all registers
Read 4 bytes @ address 0x000236A4 (Data = 0xFFA2F7FF)
```

The break function I use is:
```c
static inline void break_here(void) {

  __asm volatile("bkpt");

  __asm volatile(
      "bkpt #0x01;"
      "mov pc, lr;"
  );
}
```

I had to write this values in the debug unit to make it working with OpenOCD and St-Link V2:
```
monitor mww 0xe0042004 0x00000001
monitor mww 0xe0042008 0x00001800
```

Unfortunately it doesn't work with Segger GDB and J-Link :(

ilg-ul, 2016-04-11:
I just retested, both "bkpt" and "bkpt #0x01" halt the j-link on my STM32F4, without any special configuration.

from your messages I see you are using a nordic device; maybe j-link has some problems to configure it properly?

anyway, I doubt the problem is in the plug-ins.

metc, 2016-04-11:
Yes, I use an nRF51822 chip (Cortex M0). I had to write specific values to make it working with OpenOCD, as described in my previous message. The same C function halts de processor if I use Keil uVision as IDE. A `JLinkSettings.ini` file is generated and can be modified. Is it also possible to use a such file with the GNU ARM C/C++ J-Link Debugging plugin ?

A setting seams to be different between the GNU ARM C/C++ J-Link Debugging plugin and uVision. Sorry I am pollinating the issue tracker, but it is difficult to figure out what the problem is.

ilg-ul, 2016-04-11:
> A setting seams to be different 

you can enable the j-link log in both cases and compare the results, then compensate for the differences with additional commands.

metc, 2016-04-11:
Is the `JLinkSettings.ini` specifric to keil uVision or it is also used with the GNU ARM C/C++ J-Link Debugging plugin ?

ilg-ul, 2016-04-11:
it is specific to uVision. Eclipse stores all settings in separate debug configuration files.

metc, 2016-05-25:
Hello @ilg-ul 

I come back to you after a long time. I made several tests, and the software breakpoint works with the Segger tools version `5.10m`, but it doesn't with the latest version.

This is my setup:

- JLinkARM.dll version `V5.12f`
- Target device `nRF51822_xxAA`
- Firmware: J-Link Lite-Cortex-M V8
- GNU ARM C/C++ J-Link Debugging version `3.2.1.201604190915`
- arm-none-eabi-gcc.exe (GNU Tools for ARM Embedded Processors) version `5.3.1 20160307`

I don't no if this is an issue related to the Segger tools or the Eclipse plugin. What do you suggest ?

ilg-ul, 2016-05-25:
to be sure I get it right: in exactly the same environment, with the same project, with multiple SEGGER J-Link software installed in separate folders, if you change the path (in Preferences -> Run/Debug -> SEGGER J-Link) to 5.10m it works and with 5.12f it does not?

metc, 2016-05-25:
Yes, this is correct. In fact, I only changed the GDB executable file in the debugger configuration tab from `${jlink_path}/${jlink_gdbserver}` to the 5.10m version which is installed in a separated folder and it worked.

ilg-ul, 2016-05-25:
then I suggest to enable the J-Link log in both cases and check for differences. if you still cannot identify the problem, report the problem to SEGGER.

metc, 2016-05-25:
Please find the two log files attached:
 * [JLinkGDBServerCL_v5.12f.txt](https://github.com/gnuarmeclipse/plug-ins/files/281403/JLinkGDBServerCL_v5.12f.txt) (do not work)
 * [JLinkGDBServerCL_v5.10m.txt](https://github.com/gnuarmeclipse/plug-ins/files/281402/JLinkGDBServerCL_v5.10m.txt) (works)

ilg-ul, 2016-05-25:
no idea. 

by log I mean the detailed log, you need to enable it in the interface, and provide file names.

send them to SEGER, I'm afraid I can't be of any more help with this issue.

metc, 2016-05-25:
Thanks @ilg-ul FYI, please find attached the full logs:
 * [JLinkGDBServerCL_v5.10m-full.txt](https://github.com/gnuarmeclipse/plug-ins/files/281467/JLinkGDBServerCL_v5.10m-full.txt)
 * [JLinkGDBServerCL_v5.12f-full.txt](https://github.com/gnuarmeclipse/plug-ins/files/281466/JLinkGDBServerCL_v5.12f-full.txt)

ilg-ul, 2016-05-25:
there are obviously many differences, SEGGER should be able to figure out what is wrong.

metc, 2016-05-25:
Thanks for you help @ilg-ul . I will come back to you If the Segger guys can help me, hope so ;)

ilg-ul, 2016-05-25:
you're welcome. 

SEGGER support is good, if you provide the logs, they should be able to help.

ilg-ul, 2016-06-13:
I guess I finally identified your problem.

GDB 7.10 changed something in the way breakpoints halt execution, and current SEGGER JLinkGDBServer had problems to correctly inform GDB about the BKPT.

I provided SEGGER all details and they promised a fix with the next release.

as a temporary workaround, you can use GDB 7.8, it properly halts at breakpoints with the current SEGGER GDB server.

metc, 2016-06-14:
Hi @ilg-ul 

Thanks for your replay and the investigation. Really appreciated.
I also posted an issue on the Segger forum: http://forum.segger.com/index.php?page=Thread&postID=11183&highlight=breakpoint#post11183

I will test it again when the next stable release from Segger is available.

metc, 2016-06-30:
This is the changelog for the Segger tools  V5.12h:
> - DLL: Silicon Labs EFM8 series devices: Under special circumstances, it could happen that high-level (C-source) stepping in the IDE did not work correctly. Fixed.
> - DLL: Silicon Labs EFM8 series devices: Under special circumstances, it could happen, that breakpoints were not hit if a memory read/write request was issued by the IDE, while the CPU was running. Fixed.
> - DLL: Under special circumstances, disassembly of Cortex-AR big endian devices did not work properly. Fixed.
> - GDB Server: GDB 7.10 did not break on bkpt instruction due to wrong stop reason. Fixed.
> - J-Link Commander: J-Link command files were opened using single access instead of shared access. Fixed.

The `__BKPT(0);` instruction works now as expected on my side. @ilg-ul thank you very much for your help and support.

closed 2016-06-30 by metc

ilg-ul, 2016-06-30:
you're welcome!