eclipse-embed-cdt/eclipse-plugins #113: Strange error message on Windows if GDB server is not started locally

ErichStyger, 2016-04-13:
In case the 'start the J-Link GDB server locally is not checked, and if there is no GDB server running, there is a there is a strange error message:
----------------------------------------------------------------------------------
Error in final launch sequence
 Failed to execute MI command:
 -target-select remote localhost:2331
 Error message from debugger back end:
 localhost:2331: The system tried to join a drive to a directory on a joined drive.
 Failed to execute MI command:
 -target-select remote localhost:2331
 Error message from debugger back end:
 localhost:2331: The system tried to join a drive to a directory on a joined drive.
 localhost:2331: The system tried to join a drive to a directory on a joined drive.
----------------------------------------------------------------------------------
That message shows up at least on Windows (7 in my case).
Would be good if the error message could be improved, otherwise users have no clue what is the problem.
Something like "cannot connect to GDB server. Check that the GDB server is running or enable starting the GDB server locally in the debug configuration".

See https://mcuoneclipse.com/2016/04/13/fix-for-gdb-and-the-system-tried-to-join-a-drive-to-a-directory-on-a-joined-drive/

ilg-ul, 2016-04-13:
The message does not come from the plug-ins (even my bad English is not bad enough to imagine such an explanation!), but to me it looks like an error somewhere, so I'll investigate the cause, and try to fix it.

TM1234, 2016-04-13:
I tried to post this on Erich's blog but it didn't seem to make it... I'll try again here:

I have also seen this message and it doesn't make sense. But I noticed this:

https://msdn.microsoft.com/en-us/library/windows/desktop/ms681382%28v=vs.85%29.aspx?f=255&MSPPError=-2147217396

ERROR_JOIN_TO_JOIN
138 (0x8A)
The system tried to join a drive to a directory on a joined drive.

This error seems to crop up when the debug component sitting between GDB and the target (e.g. OpenOCD, J-Link GDB Server etc.) is misconfigured or - as above - simply not told to run.

Judging by this:

http://stackoverflow.com/questions/29747304/how-to-troubleshoot-an-error-138-error-join-to-join

the error may actually be coming from GDB and not any Eclipse component?

The error certainly doesn't make obvious sense in this context but perhaps some of this info will help to track down why it happens whatever about addressing it - especially if it's ultimately a GDB issue?

Hope this helps.

closed 2017-04-24 by ilg-ul