eclipse-embed-cdt/eclipse-plugins #114: ADuCM320 peripherals do not show up

toddrope, 2016-04-14:
I have packs and ADuCM320 installed (tried 1.1.0 and 1.0.3). The peripheral view is empty and a message shows saying "Failed to get the peripherals descriptions". When I first downloaded it would not even detect (1.1.0) until I copied a file "AnalogDevices.ADuCM320_DFP.1.1.0.pdsc". After that I could successfully select a device in the build settings, but nothing shows up in peripherals (below error). Then I downgraded to 1.0.3 with the same result (below)

The log file shows the following fault:

!ENTRY ilg.gnuarmeclipse.managedbuild.packs 4 1 2016-04-14 15:55:31.012
!MESSAGE Internal Error
!STACK 0
com.sun.org.apache.xerces.internal.impl.io.MalformedByteSequenceException: Invalid byte 1 of 1-byte UTF-8 sequence.
	at com.sun.org.apache.xerces.internal.impl.io.UTF8Reader.invalidByte(Unknown Source)
	at com.sun.org.apache.xerces.internal.impl.io.UTF8Reader.read(Unknown Source)
	at com.sun.org.apache.xerces.internal.impl.XMLEntityScanner.load(Unknown Source)
	at com.sun.org.apache.xerces.internal.impl.XMLEntityScanner.skipChar(Unknown Source)
	at com.sun.org.apache.xerces.internal.impl.XMLDocumentFragmentScannerImpl$FragmentContentDriver.next(Unknown Source)
	at com.sun.org.apache.xerces.internal.impl.XMLDocumentScannerImpl.next(Unknown Source)
	at com.sun.org.apache.xerces.internal.impl.XMLDocumentFragmentScannerImpl.scanDocument(Unknown Source)
	at com.sun.org.apache.xerces.internal.parsers.XML11Configuration.parse(Unknown Source)
	at com.sun.org.apache.xerces.internal.parsers.XML11Configuration.parse(Unknown Source)
	at com.sun.org.apache.xerces.internal.parsers.XMLParser.parse(Unknown Source)
	at com.sun.org.apache.xerces.internal.parsers.DOMParser.parse(Unknown Source)
	at com.sun.org.apache.xerces.internal.jaxp.DocumentBuilderImpl.parse(Unknown Source)
	at ilg.gnuarmeclipse.core.Xml.parseFile(Xml.java:43)
	at ilg.gnuarmeclipse.debug.gdbjtag.datamodel.SvdUtils.getTree(SvdUtils.java:201)
	at ilg.gnuarmeclipse.debug.gdbjtag.services.PeripheralsService.getPeripherals(PeripheralsService.java:170)
	at ilg.gnuarmeclipse.debug.gdbjtag.viewmodel.peripherals.PeripheralsVMNode.updateElementsInSessionThread(PeripheralsVMNode.java:258)
	at org.eclipse.cdt.dsf.ui.viewmodel.datamodel.AbstractDMVMNode$3.run(AbstractDMVMNode.java:237)
	at java.util.concurrent.Executors$RunnableAdapter.call(Unknown Source)
	at java.util.concurrent.FutureTask.run(Unknown Source)
	at java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.access$201(Unknown Source)
	at java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.run(Unknown Source)
	at java.util.concurrent.ThreadPoolExecutor.runWorker(Unknown Source)
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(Unknown Source)
	at java.lang.Thread.run(Unknown Source)

!ENTRY ilg.gnuarmeclipse.packs.data 4 1 2016-04-14 15:55:31.018
!MESSAGE Failed to get the peripherals descriptions.

ilg-ul, 2016-04-15:
aha. looks like the packs you are using contain illegal characters. if so, all I can do is to issue a more descriptive error message.

do you have a better suggestion?

toddrope, 2016-04-15:
I'm assuming you mean parsing the SVD file? Can you take a look? It's just getting it from Analog Devices pack. If I can just remove the bad character(s) then I can use... See attached.
[ADuCM320.zip](https://github.com/gnuarmeclipse/plug-ins/files/221042/ADuCM320.zip)

ilg-ul, 2016-04-15:
> parsing the SVD file?

yes, your stack trace mentions SvdUtils.getTree().

> ADuCM320.zip

you're overestimating me, I'm not that good to spot a wrong character in a 800K file :-(

toddrope, 2016-04-15:
Well I found it. Bad single quote. Working now, fantastic plugin! Any chance your descriptive error message can include a line number in the file? Or does the parser not have that information?

ilg-ul, 2016-04-15:
if the parser provides it, I'll try to display it.

ilg-ul, 2017-04-24:
display of line & column fixed in #137

closed 2017-04-26 by ilg-ul