eclipse-embed-cdt/eclipse-plugins #115: J-Link source not found for RAM breakpoint

artyeffem, 2016-05-05:
I'm running Mars.2 4.5.2 with J-Link Debugging plug-in 3.2.1 on Tiva Cortex-M.  Everything works great except when placing breakpoints on RAM functions.  I can hit the breakpoint, but get "No source available" in the view.  This is not an issue for flash breakpoints.  In addition, when setting ramfunc breakpoints, I may hit an erroneous one in the same RAM region.

ilg-ul, 2016-05-05:
thank you for your report, but I'm afraid there is not much I can do, the J-Link plug-in only starts the gdb server, the debugging is done by CDT classes.

you may try to use `-O0`, sometimes it improves the debugger behaviour.

you may also try to remove all breakpoints before starting a debug session.

artyeffem, 2016-05-06:
Thanks for the response, I wanted to add that this may be more of a compiler issue.  I started up gdb outside of Eclipse and set a couple of breakpoints, one in flash, one in RAM.  The "info b" shows both at the respective addresses, but only the flash breakpoint has a function name and source path in the "What" column.  I'll take this up with IAR.

closed 2017-04-24 by ilg-ul