eclipse-embed-cdt/eclipse-plugins #116: Unable to install GNU ARM C/C++ J-Link Debugging plugin on Neon RC1

Mardy-Marshall, 2016-05-17:
I know that it hasn't been officially released yet but thought it was worth a heads up on this.  I tried installing the latest version of gnuarmeclipse onto Neon RC1 and ran into an unresolved dependency when installing the J-Link plugin.  This is the error that Eclipse reported:

Cannot complete the install because of a conflicting dependency.
  Software being installed: GNU ARM C/C++ J-Link Debugging 3.2.1.201604190915 (ilg.gnuarmeclipse.debug.gdbjtag.jlink.feature.group 3.2.1.201604190915)
  Software currently installed: C/C++ Development Tools 9.0.0.201605031942 (org.eclipse.cdt.feature.group 9.0.0.201605031942)
  Only one of the following can be installed at once: 
    C/C++ Development Tools Core 6.0.0.201603221804 (org.eclipse.cdt.core 6.0.0.201603221804)
    C/C++ Development Tools Core 6.0.0.201605031942 (org.eclipse.cdt.core 6.0.0.201605031942)
    C/C++ Development Tools Core 5.12.0.201602011105 (org.eclipse.cdt.core 5.12.0.201602011105)
  Cannot satisfy dependency:
    From: GNU ARM C/C++ J-Link Debugging 3.2.1.201604190915 (ilg.gnuarmeclipse.debug.gdbjtag.jlink 3.2.1.201604190915)
    To: bundle org.eclipse.cdt.debug.mi.core 7.3.0
  Cannot satisfy dependency:
    From: GNU ARM C/C++ J-Link Debugging 3.2.1.201604190915 (ilg.gnuarmeclipse.debug.gdbjtag.jlink.feature.group 3.2.1.201604190915)
    To: ilg.gnuarmeclipse.debug.gdbjtag.jlink [3.2.1.201604190915]
  Cannot satisfy dependency:
    From: C/C++ Development Tools GDB/MI CDI Debugger Core 7.3.0.201602011105 (org.eclipse.cdt.debug.mi.core 7.3.0.201602011105)
    To: bundle org.eclipse.cdt.core [5.0.0,6.0.0)
  Cannot satisfy dependency:
    From: C/C++ Development Tools 9.0.0.201605031942 (org.eclipse.cdt.feature.group 9.0.0.201605031942)
    To: org.eclipse.cdt.platform.feature.group [9.0.0.201605031942]
  Cannot satisfy dependency:
    From: C/C++ Development Platform 9.0.0.201605031942 (org.eclipse.cdt.platform.feature.group 9.0.0.201605031942)
    To: org.eclipse.cdt.core [6.0.0.201605031942]

And here is the installation details for Eclipse:
  C/C++ Development Tools	9.0.0.201605031942	org.eclipse.cdt.feature.group	Eclipse CDT
    C/C++ Common GDB Support	9.0.0.201605031942	org.eclipse.cdt.gdb.feature.group	Eclipse CDT
    C/C++ Development Platform	9.0.0.201605031942	org.eclipse.cdt.platform.feature.group	Eclipse CDT
    C/C++ DSF GDB Debugger Integration	9.0.0.201605031942	org.eclipse.cdt.gnu.dsf.feature.group	Eclipse CDT
    C/C++ GNU Toolchain Build Support	9.0.0.201605031942	org.eclipse.cdt.gnu.build.feature.group	Eclipse CDT
    C/C++ GNU Toolchain Debug Support	9.0.0.201605031942	org.eclipse.cdt.gnu.debug.feature.group	Eclipse CDT
  C/C++ GCC Cross Compiler Support	9.0.0.201605031942	org.eclipse.cdt.build.crossgcc.feature.group	Eclipse CDT
  C/C++ GDB Hardware Debugging	9.0.0.201605031942	org.eclipse.cdt.debug.gdbjtag.feature.group	Eclipse CDT
  C/C++ Memory View Enhancements	9.0.0.201605031942	org.eclipse.cdt.debug.ui.memory.feature.group	Eclipse CDT
  Eclipse Platform	4.6.0.I20160512-1000	org.eclipse.platform.ide	Eclipse.org
    Eclipse Platform	4.6.0.v20160512-1000	org.eclipse.platform.feature.group	Eclipse.org
    Equinox p2, Provisioning for IDEs.	2.2.200.v20160504-1450	org.eclipse.equinox.p2.user.ui.feature.group	Eclipse.org - Equinox
  GNU ARM C/C++ Cross Compiler	2.2.1.201604190915	ilg.gnuarmeclipse.managedbuild.cross.feature.group	Liviu Ionescu

ilg-ul, 2016-05-17:
thank you, I'll investigate, but a solution will be considered after the official version is released.

mvrhov, 2016-05-20:
That's a bit unfortunate. Because only Neon works in latest ubuntu 16.04 without hogging cpu

ilg-ul, 2016-05-20:
on linux I know of some GTK problems, did you try to switch back to an older GTK?

mvrhov, 2016-05-20:
I made a fresh install of ubuntu 16.04. So there is nothing to revert to. BTW it seems that there is the same problem with OpenOCD

ilg-ul, 2016-05-20:
> there is nothing to revert to

I don't know of ubuntu 16, but most other linux distros required to manually configure some environment variable to use GTK2 instead of GTK3 for Eclipse to work properly. google it for more details.

instead of using Neon beta, perhaps a Mars 2 with this fix would be more stable.

Mardy-Marshall, 2016-05-20:
The reason that I am anxious to move to Neon is for HiDPI display support.  Mars 2 does not properly scale the UI on high resolution displays making it unusable.

ilg-ul, 2016-05-20:
right, it seems a good reason. personally I have no experience with HiDPI displays.

I guess Neon will be out in a few weeks, we'll see then what problems we have.

ilg-ul, 2016-05-20:
according to my tests, I confirm that on Neon the debugging plug-ins fail to install.

the problem is due to a dependency to a deprecated plug-in (org.eclipse.cdt.debug.mi.core).

the solution is to rewrite the debugging plug-ins to avoid using the deprecated plug-in. 

I do not know yet how complicated this will be. time permitting, I'll give it a try next week.

jonahkichwacoders, 2016-05-24:
I have added Pull 119 that allows the plug-ins to build against Neon.

ilg-ul, 2016-05-26:
@mvrhov, Miha, can you install the CDT 9.0 RC3 beta from https://hudson.eclipse.org/cdt/job/cdt-9.0/lastSuccessfulBuild/artifact/releng/org.eclipse.cdt.repo/target/org.eclipse.cdt.repo.zip, and then the new beta version of the plug-ins from updates-test?

there was a long discussion on the cdt-dev list, and finally Jonah came with a solution (see #120) which seems ok, but one additional confirmation would be helpful.

jonahkichwacoders, 2016-05-26:
Please note that the link is **not** CDT 9.0 RC3 but rather a simple build that contains the fix that will also be in RC3. It can be used to test if the changes I made to CDT and GNU ARM Plug-ins work well together for Neon.

CDT RC3  will be released in due course, and if there is still a problem with what has been done to resolve the GNU ARM plug-ins for you there is time to fix it before the release so that everyone gets the best experience.

Thank you to everyone that has taken the time to report the problem and to test out pre-releases so that the community as a whole can benefit.

mvrhov, 2016-05-27:
I've installed both over M7 neon and it seems to install without problems.. However I haven't tried to debug.

ilg-ul, 2016-05-27:
great, thank you.

as Jonah pointed out, the archive contained his own build. I guess the fix will find its way to the official repo, but, to be sure, please run again your test with the official CDT 9.0 RC3 when available and let us know if any problems.

ZahlGraf, 2016-05-27:
Hi,

I tried today the solution with the unofficial build from above and can confirm, that installing and debugging works like a charm. Thanks for your fast response to that issue :+1: 

Best regards

ilg-ul, 2016-05-27:
most of the credits should go to Jonah, I was just noisy enough for the CDT team to acknowledge the problem ;-)

ilg-ul, 2016-05-31:
for the records, the 3.1 beta plug-ins were successfully tested with [Neon RC3](http://download.eclipse.org/tools/cdt/builds/neon/milestones/rc3).

FivazTimothee, 2016-11-23:
Hello, thank you for all the work ! I'm currently facing issues while trying to perform an _offline_ installation of the GNUARMEclipse plugin on Eclipse CDT Neon.1. I've correctly downloaded the archive (ilg.gnuarmeclipse.repository-3.1.1-201606210758.zip ) from the [releases](http://gnuarmeclipse.github.io/developer/releases/) page, and when I want to install it on my Eclipse Neon CDT I face the following error : 

> Cannot complete the install because one or more required items could not be found.
  Software being installed: GNU ARM C/C++ J-Link Debugging 4.1.1.201606210758 (ilg.gnuarmeclipse.debug.gdbjtag.jlink.feature.group 4.1.1.201606210758)
  Missing requirement: GNU ARM C/C++ Debugging GDB 3.1.1.201606210758 (ilg.gnuarmeclipse.debug.gdbjtag 3.1.1.201606210758) requires 'bundle org.eclipse.cdt.debug.gdbjtag.core 8.2.0' but it could not be found
  Cannot satisfy dependency:
    From: GNU ARM C/C++ J-Link Debugging 4.1.1.201606210758 (ilg.gnuarmeclipse.debug.gdbjtag.jlink 4.1.1.201606210758)
    To: bundle ilg.gnuarmeclipse.debug.gdbjtag 0.0.0
  Cannot satisfy dependency:
    From: GNU ARM C/C++ J-Link Debugging 4.1.1.201606210758 (ilg.gnuarmeclipse.debug.gdbjtag.jlink.feature.group 4.1.1.201606210758)
    To: ilg.gnuarmeclipse.debug.gdbjtag.jlink [4.1.1.201606210758]

The version of Eclipse I have is a clean new one. It has CDT 9.10.201609121658 and the Eclipse core 4.6.1.20161007-1200. As mentionned before this is an offline installation. Was the solution to this issue involving other dependencies than the one already in CDT ? 

Thank you again for your amazing work ! I've been using your plugin for a while and it really helps !

ilg-ul, 2016-11-23:
the debugging plug-ins require the **GDB Hardware Debugging** plug-in. during online installs, it is automatically downloaded, but for offline configurations you have to manually install it.

FivazTimothee, 2016-11-24:
Oh that's right ! Sorry for not having seen it in your "How to install" page, and thank you for the help :-)

closed 2016-11-24 by ilg-ul