eclipse-embed-cdt/eclipse-plugins #117: Add linker script dependency to build makefiles

johnmoule, 2016-05-24:
Hi Liviu,

This issue applies to Executable projects which have the CDT property "Generate Makefiles automatically" checked.

When modifying source files and doing a build an existing binary file will be re-linked. This is the normal behaviour. However, modifications to the linker script(s) do not cause a re-link. You have to do a clean and then build to get a new binary file.

This feature request is to consider adding linker script dependencies into the generated makefiles.

Generally the makefile has the following dependency for elf files:

hello.elf: $(OBJS) $(USER_OBJS)
The OBJS variable is created by the subdir.mk files for each of the source directories which are all included by the main makefile (they each append the list of object files in that directory).

The USER_OBJS variable is defined in the top-level objects.mk file (based on the "Other objects" field in the linker settings in project properties).

Various other variables are defined in the top-level makefile. So this could define a variable with the linker script (or scripts) and add it to the dependency line. For example:

LINKER_SCRIPTS := ../cmsis/MK64FN1M0VDC12.ld
...
hello.elf: $(OBJS) $(USER_OBJS) $(LINKER_SCRIPTS)

One workaround for this is to create a file in the top-level project directory called makefile.defs (this is included in the makefile, if present) defining the required linker script dependency, for example:

hello.elf: ../cmsis/MK64FN1M0VDC12.ld

ilg-ul, 2016-05-24:
> consider adding linker script dependencies into the generated makefiles

agree.

but how do you suggest to implement it? the last time when I checked, the CDT classes used to generate the makefiles were not derivable, and copying them with all dependencies in the plug-ins is not a solution.

johnmoule, 2016-05-24:
Not sure exactly, but I was expecting you could use <extension point="org.eclipse.cdt.managedbuilder.core.buildDefinitions"> and specify your own makefilegenerator in <toolChain>.<build>.<buildfileGenerator> - "Allows you to supply a custom build file generator that conforms to the IManagedBuilderMakefileGenerator interface."

Looks like you could override org.eclipse.cdt.managedbuilder.makegen.gnu.GnuMakefileGenerator.

Or perhaps there's something useful in:

  org.eclipse.cdt.managedbuilder.makegen.gnu.GnuMakefileGenerator.addRuleForSource(String, StringBuffer, IResource, IPath, IResourceInfo, boolean, Vector<IPath>, Vector<IPath>)

  or

  <tool>.<inputType>.<dependencyCalculator>

Maybe with one of the above you can define linker files with extensions *.ld to be included in the build.

ilg-ul, 2016-05-25:
> Looks like you could override org.eclipse.cdt.managedbuilder.makegen.gnu.GnuMakefileGenerator.

yes, my understanding was that I have to rewrite it entirely.

if you can further experiment with the possible solutions, it'll make things faster.

johnmoule, 2016-05-25:
OK, I'll try and have a look. Won't be for a few weeks though unfortunately.