eclipse-embed-cdt/eclipse-plugins #122: "Build selected file(s)" generates list file with wrong name

jan-r, 2016-06-24:
V2.11.3, Eclipse 4.5.2

My project uses the external builder. However, building a single file (editor context menu, "Build selected file(s)") invokes the internal builder. If list file generation is activated, the command line passed to the compiler will contain the option -Wa,-adhlns=$@.lst. This works with the external builder, because make resolves $@ to the current object file's name (with correct relative path to the configuration's build directory). For a single file, this resolution does not work, so the compiler will output a file $@.lst in the configuration's build directory. 

This is quite confusing, because list files from the last full build may still exist, and you may find yourself looking at an old file, wondering why it doesn't contain your latest changes.

ilg-ul, 2016-06-24:
> For a single file, this resolution does not work

yes, this most probably uses the internal builder, which does not support all external builder options.

I have little control on the internal builder.

jan-r, 2016-06-24:
Wouldn't it be possible to explicitly use the external builder for single files? The generated makefiles contain targets for each single object, so on the command line you can build single files with something like "make src/myfile.o". I know that this must be possible with Eclipse (because e.g. TI's Code Composer Studio compiles single files this way), but I don't know how deeply you have to dig into Eclipse to extend it like this.

ilg-ul, 2016-06-24:
> Wouldn't it be possible to explicitly use the external builder for single files? 

I do not know. but if you provide a patch, I can consider it.

jan-r, 2016-06-24:
I can provide a workaround, based on this information: https://www.lpcware.com/content/forum/build-selected-file-question

- Remove the option "Generate assembler listing (-Wa,-adhlns="$@.lst")"
- Under "Other compiler flags", add -Wa,-adhlns="${OutputFileRelPath}.lst"

As this is using a build macro and not a make variable, it also works with the internal builder. And the external builder seems to translate ${OutputFileRelPath} into $@, so the generated makefiles are still the same.

One side note: Can you confirm that "Clean project" does not remove the .lst files, neither with the old method, nor with my workaround? If I interpret the makefiles correctly, the "clean" target does not add those to the rm command line.

ilg-ul, 2016-06-24:
> -Wa,-adhlns="${OutputFileRelPath}.lst"

I'll investigate if this syntax is ok instead of the current one

> Can you confirm that "Clean project" does not remove the .lst files

that's correct, list and map files are not removed

ilg-ul, 2017-04-24:
I checked building selected files with Neon.3 and it seems ok now.