eclipse-embed-cdt/eclipse-plugins #123: How to manually edited .idx file and create a local repository in Eclipse, so the needed packs can be downloaded

dili-dili, 2016-06-27:
Hi, 

I have carefully followed Mr. Lonescu's tutorial (http://gnuarmeclipse.github.io/plugins/install/) to install the gnu arm eclipse plug ins, for about 7 times. I have been stuck downloading the packs for a whole day.

I am using Eclipse mars 2 on mac El Captian 10.11.5.

I have tried both repositories(http://gnuarmeclipse.sourceforge.net/updates, and http://gnuarmeclipse.sourceforge.net/updates-test), both return the following errors.

> Processing "/Users/dili/Documents/Packages/.cache/NordicSemiconductor.nRF_SoftDevice_S310.3.0.1.pdsc" for content.xml ...
Schema version "1.2".
Completed in 278ms.
"The element type "description" must be terminated by the matching end-tag "</description>".", ignored.
Copy 517B
 from "http://downloads.sourceforge.net/project/gnuarmeclipse/xPacks/packages/content.xml?r=&ts=1467020478&use_mirror=pilotfiber"
 redirected from "http://gnuarmeclipse.sourceforge.net/packages/content.xml"
 to   "/Users/dili/Documents/Packages/.cache/.content_gnuarmeclipse_sourceforge_net_packages.xml"
Update packs completed in 289s.

> 2016-06-27 11:41:19
> Loading repos summaries...
> Parsing cached content file "/Users/dili/Documents/Packages/.cache/.content_www_keil_com_pack.xml"...
> File does not exist, ignored.
> Parsing cached content file "/Users/dili/Documents/Packages/.cache/.content_gnuarmeclipse_sourceforge_net_packages.xml"...
> File parsed in 1ms.
> Identifying installed packages...
> Found no installed packages.
> Completed in 2ms.

I found a workaround after searching, which modifies the keil.index.idx file and creates a local repository, but I am not able to reenact this based on the description. I tried to create a local repository at Preferences->packages, but doesn't work.

Could anyone tell me, where and how to create this local repository? Many thanks!

ilg-ul, 2016-06-27:
> I have been stuck downloading the packs for a whole day.

unfortunately some of the packages are not valid, and there is not much the regular user can do, but ignore them.

> Could anyone tell me ...

for general questions and discussions, please use the project forum.

dili-dili, 2016-06-27:
Hi Mr. Lonescu, 
Thanks for your reply and please forgive my recklessness of asking question here.

best wishes

Di

closed 2017-04-24 by ilg-ul