eclipse-embed-cdt/eclipse-plugins #125: Project properties Tools Paths does not support eclipse variables

johnmoule, 2016-07-08:
The toolchain path can be set globally, or for the workspace, or for a single project.
The global and workspace settings (Preferences > C/C++ > Build > Global Tools Paths and Workspace Tools Paths) can accept eclipse variables*; eg ${eclipse_home}bin.

This is not the case for the project settings (Project properties > C/C++ Build > Tools Paths). When an eclipse variable is entered into either the "Build tools folder" or the "Toolchain folder" the OK and Apply buttons are disabled and the message "Value must be an existing directory" is displayed at the top of the dialog.

*: the ability to accept eclipse variables in paths, or a currently non-existent path is provided by the following defaults:
  http://gnuarmeclipse.github.io/developer/preferences/ilg.gnuarmeclipse.managedbuild.cross/#defaults

### Steps to Reproduce

1. In a C Executable Project with toolchain set to Cross ARM GCC open project properties  > C/C++ Build > Tools Paths

2. In "Build tools folder" or "Toolchain folder" field enter a path containing an eclipse variable (eg ${eclipse_home}bin) and Click OK.
  Actual: properties dialog is not dismissed. Instead the OK and Apply buttons become greyed out.
  Expected: properties dialog is dismissed and value is set.

3. Click in the "Build tools folder" or "Toolchain folder" field again.
  Actual: Page message area displays error marker and  text "Value must be an existing directory".
  Expected: not error message.

### Versions

* [plug-in version] tag v3.1.1-201606210758, ilg.gnuarmeclipse.managedbuild.cross v2.2.1
* [Eclipse version] Mars.2
* [Java version]  1.7
* [operating system] All

I can supply a patch if you like?

ilg-ul, 2016-07-08:
> I can supply a patch if you like?

yes, please do.

johnmoule, 2017-01-12:
Applying updates from http://gnuarmeclipse.sourceforge.net/updates-test (201701102234) provides fixes for:

1) Preferences Global Tools Paths and Workspace Tools Paths don't honour strict folder check #126
2) Project properties Tools Paths does not support eclipse variables #125
3) Unable to start a debug session after updating to Eclipse Neon.2 (4.6.2) #156
   I've tested with J-Link debug adapter only.

ilg-ul, 2017-01-14:
fixed in v3.2.1-201701141320

closed 2017-01-14 by ilg-ul