eclipse-embed-cdt/eclipse-plugins #126: Preferences Global Tools Paths and Workspace Tools Paths don't honour strict folder check

johnmoule, 2016-07-08:
The global and workspace settings (Preferences > C/C++ > Build > Global Tools Paths and Workspace Tools Paths) can be configured to either accept or reject non existent folders using the preference setting defaults specified in http://gnuarmeclipse.github.io/developer/preferences/ilg.gnuarmeclipse.managedbuild.cross/#defaults .

However, setting the above defaults to true does not enable the folder check; non existent folders can be set.

### Steps to Reproduce

0. Create and launch an Eclipse product with a preferenceCustomization file (eg: "plugin_customization.ini") using extension point org.eclipse.core.runtime.products containing the following values:

  ilg.gnuarmeclipse.managedbuild.cross/global.toolchain.path.strict=true
  ilg.gnuarmeclipse.managedbuild.cross/global.buildTools.path.strict=true
  ilg.gnuarmeclipse.managedbuild.cross/workspace.toolchain.path.strict=true
  ilg.gnuarmeclipse.managedbuild.cross/workspace.buildTools.path.strict=true

1. Open Preferences > C/C++ > Build > Global Tools Paths

2. In "Build tools folder" or "Toolchain folder" field enter a path that does not exist and Click OK.
  Actual: properties dialog is dismissed and value is set.
  Expected: properties dialog is not dismissed. Instead the OK and Apply buttons become greyed out.

3. Click in the "Build tools folder" or "Toolchain folder" field again.
  Actual: Page message area displays no error message.
  Expected: Page message area displays error marker and  text "Value must be an existing directory".

The same happens with Preferences > C/C++ > Build > Workspace Tools Paths

### Versions

* [plug-in version] tag v3.1.1-201606210758, ilg.gnuarmeclipse.managedbuild.cross v2.2.1, ilg.gnuarmeclipse.core v3.1.1
* [Eclipse version] Mars.2
* [Java version]  1.7
* [operating system] All

related:
  Project properties Tools Paths does not support eclipse variables #125
  https://github.com/gnuarmeclipse/plug-ins/issues/125

I can supply a patch if you like?

ilg-ul, 2016-07-08:
> I can supply a patch if you like?

yes, please do.

johnmoule, 2016-07-11:
ok.
Do you have a preference on how to provide the patch. If not I'll follow this guide:
https://guides.github.com/activities/contributing-to-open-source/#contributing

ilg-ul, 2016-07-11:
yes, fork, branch, pull request should be fine.

ilg-ul, 2017-01-09:
Hi John,

I finally applied the patches. Please check the version from updates-test and let me know if it is ok.

johnmoule, 2017-01-11:
Thanks. It may be a while before I can get round to testing this.

johnmoule, 2017-01-11:
Just catching up:
https://dev.eclipse.org/mhonarc/lists/cdt-dev/msg31542.html

That's great news!

ilg-ul, 2017-01-11:
indeed!

you can test the new version from updates-test. I'll prepare a new release shortly.

johnmoule, 2017-01-12:
Applying updates from http://gnuarmeclipse.sourceforge.net/updates-test (201701102234) provides fixes for:

1) Preferences Global Tools Paths and Workspace Tools Paths don't honour strict folder check #126
2) Project properties Tools Paths does not support eclipse variables #125
3) Unable to start a debug session after updating to Eclipse Neon.2 (4.6.2) #156
   I've tested with J-Link debug adapter only.

ilg-ul, 2017-01-12:
great, thank you!

ilg-ul, 2017-01-14:
fixed in v3.2.1-201701141320

closed 2017-01-14 by ilg-ul