eclipse-embed-cdt/eclipse-plugins #129: Unnable to download package

marcosfg, 2016-07-15:
### Description

Can't download any pack.

After clicking install the process starts, however it stays at 0% and does not move. If I press cancel the process does not disappear. 

I have tried many times, using different versions of eclipse, OS, plug-in version and they all give the same error.

### Steps to Reproduce

1. Select a pack
2. Press the install button

### Versions

* [plug-in version]
3.1.1
* [Eclipse version]
Luna,Neon
* [Java version]
1.8.0.91
* [operating system]
Ubuntu 16.04 LTS, Windows 10
* [toolchain version
arm-none-eabi-gcc-4.8.3

ilg-ul, 2016-07-15:
could not reproduce this:

```
2016-07-15 18:43:27
Installing packs...
Install "Keil.STM32F4xx_DFP.2.8.0.pack".
Copy 392MB
 from "http://www.keil.com/pack/Keil.STM32F4xx_DFP.2.8.0.pack"
 to   "/Users/ilg/My Files/MacBookPro Projects/GNU ARM Eclipse/Eclipse Workspaces/Packages/.cache/Keil.STM32F4xx_DFP.2.8.0.pack"
Unzip "/Users/ilg/My Files/MacBookPro Projects/GNU ARM Eclipse/Eclipse Workspaces/Packages/.cache/Keil.STM32F4xx_DFP.2.8.0.pack".
Write "/Users/ilg/My Files/MacBookPro Projects/GNU ARM Eclipse/Eclipse Workspaces/Packages/Keil/STM32F4xx_DFP/2.8.0/Keil.STM32F4xx_DFP.pdsc".
...
Write "/Users/ilg/My Files/MacBookPro Projects/GNU ARM Eclipse/Eclipse Workspaces/Packages/Keil/STM32F4xx_DFP/2.8.0/_htmresc/STM32Cube_components.bmp".
Write "/Users/ilg/My Files/MacBookPro Projects/GNU ARM Eclipse/Eclipse Workspaces/Packages/Keil/STM32F4xx_DFP/2.8.0/_htmresc/st_logo.png".
27042 files written, 1356MB.
All files write protected.
1 pack installed.
Install completed in 105s.
```

marcosfg, 2016-07-15:
I have tried to download several LPC packs.

Is there anything I can do to debug the problem?

I have attached a screenshot from my eclipse windows and terminal output.

[image]

ilg-ul, 2016-07-15:
> Is there anything I can do to debug the problem?

you can check 
- the Packs Manager console
- the Eclipse log

ilg-ul, 2016-07-15:
btw, what gave you the certitude you identified a bug? for general questions please use the project forum.

closed 2017-04-24 by ilg-ul

Disgust, 2018-05-06:
First time I used repo file with .idx extention.
Now Ive changed it to .pidx and waiting while it downloaded (yesterday it took about 4 hours)

ilg-ul, 2018-05-06:
If the Keil server is down, there is not much we can do.

Disgust, 2018-05-06:
How to know actual path from packages should be downloaded?

ilg-ul, 2018-05-06:
From the index file.

Disgust, 2018-05-06:
Thanks and sorry for disturbing.
In my case the issue is result of Russian goverment policy and blocking of lots of IPs.
Now everything is worked.

ilg-ul, 2018-05-06:
ok, good luck.