eclipse-embed-cdt/eclipse-plugins #130: Option to load SVD files for devices that do not have packs

tazbox, 2016-07-16:
Hi Liviu,

I would like to congratulate you for developing and maintaining this plug-in. It's a big help for hobbyists worldwide.

I have an improvement suggestion related to Peripherals view. As not all silicon vendors publish CMSIS packs it would be nice to be able to load one or more SVD files (describing the core registers and the peripherals registers) even there is no device selected in C/C++ Build->Settings->Devices.

I'm willing to test this feature if you decide to implement it.

Kind regards,
Taz

ilg-ul, 2016-07-16:
thank you @tazbox, I am considering a solution for this problem, but it'll take a while to be implemented.

ilg-ul, 2018-01-05:
fixed in December 2017.

ilg-ul, 2018-01-09:
included in v4.3.1-201801092051.

closed 2018-01-09 by ilg-ul