eclipse-embed-cdt/eclipse-plugins #134: parameter checking problem of debug.gdbjtag.jlink

noodlefighter, 2016-08-08:
### Description
While debug, i want to startup the JLink GBD Server manually, but a little problem happened.

### Steps to Reproduce
0. "Debug Configurations" - "GDB SEGGER J-Link Debugging" - 
1. found the checkbox "Debugger" - "J-Link GDB Server Setup" -  "Start the J-Link GDB server locally" and uncheck it.
2. make sure there is _no text in the textbox "Device Name"_.
3. Launch Debugging

**Expected behavior:** [What you expected to happen]
Error hint:
```
Missing mandatory device name. Fill-in the 'Device name:' field in the Debugger tab.
```

### Versions
* debug.gdbjtag.jlink 4.1.1.201606210758
* eclipse version: neon release (4.6.0)
* jre version: jre-8u102-windows-x64
* Windows 7 Ultimate
* GNU Tools ARM Embedded 4.9 2015q2

ilg-ul, 2016-08-12:
thank you, I'll investigate and fix it.

ilg-ul, 2016-08-18:
please try the beta 3.2.1-201608181714 from updates-test and let me know if the problem was solved.

closed 2017-04-24 by ilg-ul