eclipse-embed-cdt/eclipse-plugins #135: Build only generates .bin file after project clean

vortex314, 2016-08-25:
### Description

When doing a first time build or a build after "project clean", the console shows that the last step of converting .elf to .bin is executed. If however I do an incremental build , the console shows that the .elf is generated but the last step of converting .elf to .bin is not done.

### Context
- Windows 7
- Eclipse Mars
- latest version of GNU ARM Eclipse
- Internal builder configured

ilg-ul, 2016-08-25:
> Internal builder configured

the CDT internal builder is no longer supported and should not be used.

vortex314, 2016-08-25:
Thanks for the swift response, I'll try with external builder, requiring installation of make.

closed 2016-08-25 by vortex314

ilg-ul, 2016-08-25:
http://gnuarmeclipse.github.io/windows-build-tools/

vortex314, 2016-08-25:
You guys, thought about everything !  Amazing work. Please continue.