eclipse-embed-cdt/eclipse-plugins #136: Eclipse marketplace install broken

andersm, 2016-08-30:
Trying to install the plugin via the Eclipse marketplace fails with the following error message:

> Unable to read repository at http://gnuarmeclipse.sourceforge.net/updates/content.xml.
sun.security.validator.ValidatorException: PKIX path building failed: sun.security.provider.certpath.SunCertPathBuilderException: unable to find valid certification path to requested target

Eclipse Neon on Windows 7 64-bit, JRE 1.8.0_74

ilg-ul, 2016-08-30:
I just installed the plug-ins on a new Eclipse via the marketplace, so I cannot confirm your report.

If you still encounter problems, it might be related to your network setup.

enz-ml, 2016-11-06:
I get the exact same error massage if I try the "Check For Updates" function in Eclipse (Mars.2 (4.5.2), Win 10 Pro 64bit; Java: Build 1.8.0_111-b14).
I use the update function approx. once per month and never got this error.
If it would be a network setup error, what could that be in detail? Anything to check?

ilg-ul, 2016-11-06:
if the problem is due to some SourceForge out-of-sync mirrors, I doubt there is something we can do.

try the beta version from updates-test, I pushed it a few days ago, maybe the mirrors are in sync.

enz-ml, 2016-11-06:
Thanks for the quick response.
I took the chance to made a fresh install with Eclipse Neo, it seems mature enough now.
I downloaded and installed the archiv manually, that worked. Any chance to get rid of sourceforge completely?

enz-ml, 2016-11-06:
Strange, downloading the Packs fails with the same error message. Are they also hosted on sourceforge?

ilg-ul, 2016-11-06:
do you mean the updates-test has the same problem?

> Any chance to get rid of sourceforge completely?

hard to tell, GitHub does not have a binary distribution mechanism, and also does not provide any statistics, which is very difficult to accept.

ilg-ul, 2016-11-06:
> downloading the Packs fails with the same error message. Are they also hosted on sourceforge?

no, they are not. you might have a different problem.

enz-ml, 2016-11-06:
Yes, The updates-test also fails with the same error message.
My internet connection is working fine, any hint?
It is just a few weeks ago that I installed Eclipe Mars + gnuarmeclipse + packs on another machine just fine.

ilg-ul, 2016-11-06:
I would install a separate instance of Eclipse CDT, create a new workspace and, without any other plug-ins, I would try to install the GNU ARM Eclipse plug-ins from updates-test.

enz-ml, 2016-11-07:
Sorry for the delay, was out of office today.

> I would install a separate instance of Eclipse CDT, create a new workspace and, without any other plug-ins, I would try to install the GNU ARM Eclipse plug-ins from updates-test. 

I just did exactly that, with MARS.2.
Same error:
Unable to read repository at http://gnuarmeclipse.sourceforge.net/updates-test/content.xml.
Unable to read repository at http://gnuarmeclipse.sourceforge.net/updates-test/content.xml.
sun.security.validator.ValidatorException: PKIX path building failed: sun.security.provider.certpath.SunCertPathBuilderException: unable to find valid certification path to requested target

I tried it also with Eclipse Neo: same result.

ilg-ul, 2016-11-07:
in a way the message is normal, the `content.xml` file **is** missing, because it is no longer used, it was replaced with a compressed version of it. but Eclipse is able to retry with the compressed url.

looking closely to your error message, it seems you have a SSL problem with your Java.

I suggest you google for the error message, perhaps you find a way to fix it. 

the repository is ok, I just tried it a few minutes ago. if you have a separate machine, or preferably a fresh virtual machine, it'll probably work.

gustavovelascoh, 2016-11-15:
HI @enz-ml, I have the same problem you describe in one machine, but in other two machines installation was successful. Did you find the cause/solution of this issue?

enz-ml, 2016-11-20:
Yes, I solved the problem finally, but it was a bit fiddly with some trial and error and I don't know if my solution works for you.

Here are my findings:

1. It didn't matter what type of update mechanism I tried (Marketplace, update site, etc.), neither worked.

2. Googling the error message didn't lead to something. All results were mentioning that the server uses a self certification, totally out of my knowledge and als out my influence.

3. Trying a virtual machine (Linux Mint, Lubuntu): Showed the same error!

4. Trying it on a different hardware: It worked! So something must be wrong with my machine.

5. Totally removing and reinstalling JAVA didnt't change anything.

6. Disabling Antivirus (Kaspersky Internet Security 2016) also didn't change anything.

7. BUT: Totally Removing AV solved the problem!

I deinstalled Kaspersky and switched to Bitdefender and things are still working.
So it seems, the problem was related to Kaspersky Antivirus together with the self certification of the server.
I use Kaspersky for nearly a year and never had this problem. I don't really know what changed.
Strange also, other plugins (bracketeer, plantuml, etc. ) and also the main eclipse site always worked. The problem only occured with gnuarmeclipse.

I hope my findings will be helpfull for others.
Maybe there is a chance to find the real cause of this.

ilg-ul, 2016-11-20:
> Strange also, other plugins (bracketeer, plantuml, etc. ) and also the main eclipse site always worked. The problem only occured with gnuarmeclipse.

the gnuarmeclipse update site does not make use of any security features at all, it is a plain folder on one of sourceforge servers. so please allow me to doubt that the problem has anything to do with gnuarmeclipse itself, I would suspect the sourceforge servers.

> Maybe there is a chance to find the real cause of this.

here is my theory: sourceforge uses some unusual certificates on their servers; Eclipse detects that something is wrong, and tries to check the certificates more thoroughly, but the AV blocks these packets and the install fails.

> I use Kaspersky for nearly a year and never had this problem.

I never ever had any antivirus on my mac, and never had any problems, neither with viruses or blocked ports.

ilg-ul, 2016-12-29:
http://gnuarmeclipse.github.io/blog/2016/12/02/plugins-install-issue/

closed 2016-12-29 by ilg-ul