eclipse-embed-cdt/eclipse-plugins #138: Working with CMake-based projects – Info

offa, 2016-09-09:
### This is an *Information* – not a bug report 
--------------------------

Using **CMake** based projects with this plugin **_works_** fine out-of-the-box for **_building_**, but **_debugging_** (eg. on HW with openocd or QEMU) **_wont work_**. Executing a debug / run configuration will crash openocd / qemu.

Since all settings like Toolchain, Flags etc. are set by Cmake, Eclipse doesn't have tool settings available (#5).

-------------------------------
## Here's how to setup a (working) CMake Project

### Basic setup

As for all CMake projects used with Eclipse:

1. **Setup** a the Cmake Project as usual (for ARM you may use a toolchain file)
1. **Import** the project into Eclipse by creating a new Project from existing Makefile
1. If necessary change **_build directory_** in Eclipse (**`Project Settings -> C/C++ Build `**, edit _"Build directory"_)

You can now **_build_** the project.

### Fix Debugging

As stated above, compilation works – debugging not.
This happens because the settings as for default C/C++ projects aren't available. Therefore the Eclipse launch configurations will use the wrong (= default system) debugger instead of the ARM one. This will kill any execution on HW and QEMU.

GDB used by run configurations: `${cross_prefix}gdb${cross_suffix}`

Since both `cross_prefix` and `cross_suffix` aren't available for this project type, the GDB executable used will expand to just `gdb`.

#### Solutions

1. Manually **_change_** the **_GDB executable_** for each **_Debug / Run configuration_** to eg. `arm-none-eabi-gdb`
1. **OR:** Go to **`Project settings -> C/C++ Build -> Build Variables`** and add `cross_prefix` = `arm-none-eabi-`<sup>(Hint)</sup> (if necessary `cross_suffix` too) – this will expand the correct executable for *all* launch configuration of that project

<sup>(Hint)</sup> _Don't forget the trailing '-'!_

-------------------------------

Whatever solution is used, building & debugging works fine using CMake and Eclipse ARM! :+1:

ilg-ul, 2016-09-09:
thank you for the info, but the GNU ARM Eclipse plug-ins focus on managed CDT projects, while CMake projects are not-managed, thus not the object of these plug-ins.

using the GNU ARM Eclipse debugging plug-ins with non-managed project is also not really supported; you can find workarounds like manually editing various fields, but you are on your own.

offa, 2016-09-09:
> thank you for the info, but the GNU ARM Eclipse plug-ins focus on managed CDT projects, while CMake projects are not-managed, thus not the object of these plug-ins.

I know, I just wanted to let you know, that they can be used to.

closed 2016-09-09 by offa

lorcap, 2018-07-26:
Here it is my approach:

* From a working CMake project, generate a ***Debug / Run Configuration*** and export it to a file such as ```debug.launch.in``` 
* Replace user-configurable information with CMake variable placeholders, e.g. ```@CMAKE_BUILD_TYPE@``` (more below)
* Use CMake command ```configure_file()``` to replace placeholders.

Currently, I found the following lines containing information that needs to be replaced as to make debugging work:

* ```<stringAttribute key="org.eclipse.cdt.dsf.gdb.DEBUG_NAME" value="@CMAKE_DEBUGGER@"/>```
* ```<stringAttribute key="org.eclipse.cdt.launch.PROJECT_ATTR" value="project-@CMAKE_BUILD_TYPE@@@CMAKE_BINARY_DIR_NAME@"/>```
* ```<listEntry value="/project-@CMAKE_BUILD_TYPE@@@CMAKE_BINARY_DIR_NAME@"/>```

Some CMake variables are available, others must be constructed. In the end, I can provide a working debug configuration with the following lines:

```CMake
find_program(CMAKE_DEBUGGER NAMES arm-none-eabi-gdb${CMAKE_EXECUTABLE_SUFFIX})
get_filename_component(CMAKE_BINARY_DIR_NAME ${CMAKE_BINARY_DIR} NAME)
configure_file(${CMAKE_SOURCE_DIR}/eclipse/debug.launch.in ${CMAKE_SOURCE_DIR}/eclipse/$debug.launch @ONLY)
```