eclipse-embed-cdt/eclipse-plugins #139: License of C/C++ code ?

bogdanm, 2016-09-10:
I can't find the license of the C/C++ code in this repository (for example the files under https://github.com/gnuarmeclipse/plug-ins/tree/develop/ilg.gnuarmeclipse.templates.stm/templates/stm32f4xx_exe_c_project). I want to re-distribute these files in another project (independent of Eclipse) and I don't know if I can do that, or what license to list for them if I can.

ilg-ul, 2016-09-10:
C/C++ code normally should refer to the µOS++, which is release under the MIT License.
 
so, from my point of view, you can redistribute the files in any other projects (well, maybe except North Korean nuclear bombs or similar).

legally all you have to do is to mention the µOS++ project (http://micro-os-plus.github.io) and the MIT license.

bogdanm, 2016-09-10:
Thanks! That's all I needed to know.

closed 2016-09-10 by bogdanm

ilg-ul, 2016-09-10:
btw, the code used by the current templates is getting obsolete.

for a glimpse of the future templates, you can check the demo projects from:

https://github.com/micro-os-plus/eclipse-demo-projects