eclipse-embed-cdt/eclipse-plugins #140: Can't exit Eclipse without halting Stm32f4Discovery

smitgd, 2016-09-15:
When eclipse/plug-in exits, it appears to send an empty packet to openocd that triggers a halt command to the target. This is true even when openocd is running independently and is not started or terminated by eclipse/plug-in.
See here for details:
https://www.element14.com/community/thread/56535/l/how-to-exit-eclipse-without-halting-stm32f4discovery

smitgd, 2016-09-17:
An easy work-around is to just terminate the "openocd" process seen in the Debug tab. Then when you exit Eclipse the board does not halt and keeps running. Still, not having to do this would be good.

armink, 2017-01-10:
+1

The same problem for J-Link. It's just work fine when using console tab terminate button.

BouchkatiTarek, 2017-06-29:
you can add detach event in stm32f4x.cfg (I m used to do so when connecting in hotplug mode)

```
$_TARGETNAME configure -event gdb-detach {
	resume
}
```