eclipse-embed-cdt/eclipse-plugins #141: Reset button does HALT and then Continue (for my config. of oocd)

smitgd, 2016-09-19:
### Description
Normally the reset button works as expected. However, for one debug configuration that I set up it does not. This is a debug configuration with Initial reset and Pre-run/restart disabled as well as Load executable disabled. With this configuration I can start Eclipse/gdb/openocd with a running target connected without affecting the target. This much seems to work fine. However, after some normal debugging, if I try to reset the target using the reset toolbar button, the target just halts for a very short time and then continues running. This is verified by looking at the packets received by openocd when running with -d3 logging option. 

[Description of the bug or feature]
The behavior of the reset button with my debug configuration is:
Halt
Continue   <--- if configured to continue

**Expected behavior:**

The preferred behavior of the reset button with my debug configuration is:
Halt
Reset   <--- and don't run, stay at reset vector
Set temporary breakpoint   <--- if a temp breakpoint is configured
Continue.   <--- if continue is configured

Note: A work-around is to enter this into the gdb console (after halting the target):
mon reset halt
tbreak boot_card  <-- where I normally configure a temp bp after a reset
continue         <-- run to boot_card

* [plug-in version]   4.1.1.201606210758
* [Eclipse version] Mars.2 (4.5.2)
* [Java version] openjdk version "1.8.0_102" 
* [operating system] Linux 4.7.3-100.fc23.x86_64 #1 SMP  
* [toolchain version]  arm-rtems4.12-gcc (GCC) 6.1.1 20160609

ilg-ul, 2016-09-19:
as far as I can remember, the pre-run reset is the one used to restart the board, so if you disable it, the debugger cannot reset the target.

smitgd, 2016-09-20:
>  the pre-run reset is the one used to restart the board, so if you disable it, the debugger cannot reset the target.

Yes, that's what I observe (but can reset via gdb console work-around). However, if possible the "Restart" button behavior should be independent of the pre-run reset enable and always do a reset (then halt or run, optionally set a temp bp and optionally continue depending on other pre-run/Restart selections). Looking at the plug-in code, I have no idea if this is possible.
If not possible, then maybe if pre-run/Restart is disabled then the Restart button should be disabled or at least do nothing instead of halt then continue.

ilg-ul, 2017-04-24:
I could not reproduce it, but I'll keep it open for a while