eclipse-embed-cdt/eclipse-plugins #143: specify device package when debugging

curtisembedded, 2016-10-27:
This is a feature request, not an issue report.

Currently it is only possible to specify "Device" when you create a project using the GNU Arm plugin. However when debugging the "Peripherals" tab appears regardless of whether the device was specified.

Would it be possible to enable package selection from the Peripherals tab?

ilg-ul, 2016-10-27:
can you be more specific? I have difficulties to identify exactly what you mean.

curtisembedded, 2016-10-28:
When debugging, it would be beneficial if the device could be selected in the peripherals tab so you can inspect the peripheral registers. The peripherals tab is visible even though the plug-in was not used to create the project.

ilg-ul, 2016-10-28:
> the device could be selected in the peripherals tab

the device is a major characteristic of the build configuration, and as such it is set in the C/C++ Settings page. once set there, the device field is automatically set in the j-lin & qemu debug configuration, and the peripheral registers are shown correctly.

what would be the purpose of selecting the device in the peripheral tab, if the value is already available from the build configuration?

curtisembedded, 2016-10-28:
Our company is not using the managed build configuration. However if would be useful to be able access the peripherals when debugging.

ilg-ul, 2016-10-28:
then this is another story; I'll investigate it.

ilg-ul, 2016-10-28:
> Our company is not using the managed build configuration.

the main reason for using Eclipse is the CDT indexer, which works best for managed projects. did you manage to configure the indexer properly for your non-managed projects?

curtisembedded, 2016-10-28:
The indexer seems to be working fine. I have been able to use the GNU ARM, openOCD debugging plugin for source level debugging. Access to the peripheral register definitions when debugging is a missing piece.

Thanks!

ilg-ul, 2016-10-28:
> The indexer seems to be working fine. 

are the preprocessor unused sequences of code correctly greyed? this requires the indexer to know exactly the symbols defined inside each source file, and this is generally a challenge for non-managed projects.

> openOCD debugging 

hmmm... you must be either very lucky, or must be kidding. for professional usage, openOCD is highly **not** recommended.

curtisembedded, 2016-10-28:
My client only has ST-Link2 JTAG/Linux development. AFAIK, openOCD is the only option. Personally/professionally I have a Segger J-Link and use that.

The point is that you can do source level debugging without using managed projects.