eclipse-embed-cdt/eclipse-plugins #144: J-Link launch configuration tabs (Debugger and Startup) aren't scrollable

johnmoule, 2016-10-31:
Some J-Link launch configuration tabs aren't scrollable The contents of the "Debugger" and "Startup" tabs of a J-Link launch configuration cannot be scrolled using a trackpad or mouse wheel, leading to a less than satisfactory standard user experience.

The "Main", "Source" and "Common" tabs' contents can all be scrolled as expected.

### To Reproduce (always)
1) Open an eclipse installation containing the GNU ARM C/C++ J-Link Debugging feature.
2) Open Debug Configurations dialog (Run > Debug configurations ...).
3) Open an existing or create a new GDB SEGGER J-Link Debugging configuration.
4) Resize the dialog so that the contents of the Main tab are clipped (a vertical scroll bar may appear)
5) Click the Debugger tab and place the insertion point in the Executable field.
6) Use the mouse scroll wheel or track pad to scroll the contents:
  Actual: the contents are not scrolled
  Expected: the contents are scrolled (try scrolling in the Main tab to see expected behaviour)

### Versions

* [plug-in version] 3.1.1-201606210758
* [Eclipse version] Mars.2
* [operating system] Windows 8.1, Linux Centos 6.5, MacOS El Capitan

johnmoule, 2016-10-31:
The fix is very simple.
The scrolling problem is caused by having a scrolled composite inside a scrolled composite. Inspected org.eclipse.cdt.launch.ui.CMainTab2.createControl(), where the problem isn't present, and noticed it does not implement its own scrolled composite. This is because scrolling is implemented by the Platform in org.eclipse.debug.internal.ui.launchConfigurations.LaunchConfigurationTabGroupViewer.showInstanceTabsFor().

The fix is to remove org.eclipse.swt.custom.ScrolledComposite from ilg.gnuarmeclipse.debug.gdbjtag.jlink.ui.TabDebugger.createControl() and ilg.gnuarmeclipse.debug.gdbjtag.jlink.ui.TabStartup.createControl().

Patch attached.
[issue144.patch.txt](https://github.com/gnuarmeclipse/plug-ins/files/561579/issue144.patch.txt)

ilg-ul, 2016-10-31:
thank you, I'll definitely consider your patch, but right now I'm quite busy with qemu.

johnmoule, 2016-10-31:
You're welcome.
cheers john

ilg-ul, 2017-04-24:
finally fixed since 3.4.1.201704241636

closed 2017-04-26 by ilg-ul