eclipse-embed-cdt/eclipse-plugins #146: Mac OS X: Need to specify full path to load Segger FreeRTOS Plugin

ErichStyger, 2016-11-03:
Not sure if this is an issue with Segger (I doubt) or with the GNU ARM Eclipse plugin (I think it is an issue what is set as 'current directory' for the Segger GDB server when it is lauchend by the GNU ARM Eclipse plugin?)

[Description of the bug or feature]
Under Windows, I can specify the following under the 'Other Options' to load the FreeRTOS Segger RTOS plugin:
-s -rtos GDBServer/RTOSPlugin_FreeRTOS

That works fine under Windows, and it finds the Segger plugin located in the GDBServer directory (inside the Segger installation).
The od thing is under Mac OS: here I have to specify the full path to the plugin, otherwise Segger does not find it:
-s -rtos/Applications/SEGGER/JLink_V610i/GDBServer/RTOSPlugin_FreeRTOS

Then it finds it:
SEGGER J-Link GDB Server V6.10i Command Line Version
JLinkARM.dll V6.10i (DLL compiled Oct 25 2016 19:26:54)
.....
Loading RTOS plugin: /Applications/SEGGER/JLink_V610i/GDBServer/RTOSPlugin_FreeRTOS.so...
RTOS plugin (v1.0) loaded successfully
RTOS plugin initialized successfully.

### Steps to Reproduce
In the 'Other options' of the Segger plugin settings, specify:
-s -rtos GDBServer/RTOSPlugin_FreeRTOS

**Expected behavior:** [What you expected to happen]
Segger console shows:
Loading RTOS plugin: GDBServer/RTOSPlugin_FreeRTOS.so...
RTOS plugin (v1.0) loaded successfully

**Actual behavior:** [What actually happened]
Loading RTOS plugin: /Applications/SEGGER/JLink_V610i/GDBServer/RTOSPlugin_FreeRTOS.so...
failed

### Versions
SEGGER J-Link GDB Server V6.10i Command Line Version
Mac OS X
  GNU ARM C/C++ J-Link Debugging	4.1.1.201606210758	ilg.gnuarmeclipse.debug.gdbjtag.jlink.feature.group	Liviu Ionescu

Many thanks,
Erich

ilg-ul, 2016-11-03:
> what is set as 'current directory' for the Segger GDB server

if I remember right, the 'current directory' is the Debug or Release folder.

did you try to start the JLinkGDBServer manually? possibly from different CWDs, to see how the server itself expects the path to the plug-in?

ErichStyger, 2016-11-03:
I have to try it with the command line/manually (it is not my machine, but from a student (several ones had the issue). The Segger server expects it relative to the executable, and it works that way on Windows. But not on Mac. I have to try it tomorrow on that student machine. I will ask Segger support as well.

Multumesc!

ErichStyger, 2016-11-04:
Hi Liviu,
I have received confirmation from Segger that the problem is on their side (both for Mac OS X and Linux). They plan to fix it in their next release.
I'm sorry (Imi pare rau) for the churn, and I'm closinig the the issue.

Erich

closed 2016-11-04 by ErichStyger

ilg-ul, 2016-11-04:
ok, no problem.