eclipse-embed-cdt/eclipse-plugins #147: wishlist: trace data decoding and display (SWO, ITM etc)

paulfertser, 2016-11-10:
Hello Liviu,

I've noticed there's an increased demand for support of integrated solution to view trace data (typically that's ITM messages output via SWO and captured with stlink, jlink or an additional UART or LA).

Currently OpenOCD provides trace data by:
1. Capturing with stlink or jlink and writing to a file
2. Capturing with stlink or jlink and sending the stream via Tcl RPC
3. Configuring the target and letting other tools capture and process data (useful when SWO capture is done with an LA or additional UART)

The approach #1 is of limited usefullness on windows as this non-standard OS doesn't support named fifos (and its "named pipes" can not be manipulated like ordinary files).

I think many of your users would like to have trace output visible directly in one of Eclipse tabs during debugging.

For your consideration, here's a couple of related links:
1. https://github.com/apmorton/OpenOcdTraceUtil an external utility (written in C#) that communicates with OpenOCD via Tcl RPC, processes and displays trace data realtime
2. http://git.zapb.de/swodec.git/ and http://git.zapb.de/libswo.git/ libraries to facilitate trace data decoding
3. https://github.com/Getty/p5-itm Perl library to decode and pretty-print (to a colour text console) trace data
4. https://github.com/Linaro/OpenCSD a C++ library to do decoding, doesn't support ITM yet

Have a nice day and happy hacking!

ilg-ul, 2016-11-10:
thank you, Paul.

FYI, the J-Link debugging plug-in has a separate tracing console (a separate tcp port), where the semihosting and SWO streams are  forwarded by SEGGER. 

in other words, decoding is done inside the J-Link GDB server.

if you want to help adoption, it would be useful to take a similar approach in OpenOCD too.

paulfertser, 2016-11-10:
I do not see how decoding such a complex format (ITM of different versions + ETM of different versions + optional framing) might be done inside OpenOCD, it feels like a completely separate goal better be delegated to an external tool.

Regarding semihosting, it's currently output to OpenOCD's console (and thus works in Eclipse out of the box) and the upcoming version will also be able to send to GDB (via its File-I/O mechanism which is like semihosting but generic accross all targets).

ilg-ul, 2016-11-10:
> it feels like a completely separate goal

if you know someone interested to support the development of such an Eclipse plug-in, please let me know.

paulfertser, 2016-11-10:
Liviu,

Marc Schink (author of http://git.zapb.de/swodec.git/ ) offers to build a standalone tool that would connect to OpenOCD's TCL, get the data, parse and provide it on a TCP socket. Would you integrate such a tool if he implements it? How do you plan to control which ITM channels to enable/disable?

ilg-ul, 2016-11-10:
> SEGGER J-Link has a configuration window to configure everything related to tracing, I don't have to do anything in the Eclipse plug-in.

correction, the J-LinkGDBServer has a command to set the channel, and the plug-in passes it to the server.

ilg-ul, 2016-11-18:
@paulfertser, I tried a build on macOS with `--enable-oocd_trace` and it failed.

I filed a bug report at https://sourceforge.net/p/openocd/tickets/137/

ilg-ul, 2016-11-18:
@paulfertser, I tried a build on mingw64/debian8 with --enable-oocd_trace and it failed.

I filed a bug report at https://sourceforge.net/p/openocd/tickets/141/

closed 2017-04-24 by ilg-ul