eclipse-embed-cdt/eclipse-plugins #149: LED blink STM32fF0 stucks at __DEBUG_BKPT();

deejayfrankcesco, 2016-11-23:
Dear All,

I have a simple but frustrating problem. I installed the last GNU ARM Eclipse plug-ins (v3.1.1-201606210758) following all the procedure on gnuarmeclipse site. I used it a lot with my stm32f407 and It works like a charm with debugging too. Now I bought two cheap chinese boards with STM32F030F4 to work with very simple projects, I program the core with the SWD on DiscoveryF4 board.

I tried to create a simple blink led project template (no semihosting) -> I setup the correct GPIO etc -> I configured openOCD debug -> It starts, program + verify -> then It stucks at "__DEBUG_BKPT();" maybe for an Hard Fault Handler. 

I can not avoid it stops. I tried also to program with release configuration and program the hex from stlink but there are no GPIO changes. I checked that clock etc. are correct. I tried also to program a compiled hex which I have got from the past and It works perfectly so I think that It's a software problem maybe in the project template.

 What should I do?

Thank you all

ilg-ul, 2016-11-23:
is this a bug report?

for questions and general discussions, please use the project forum.

deejayfrankcesco, 2016-11-23:
I think that It could be an error in the STM32F0 project template from gnuarmeclipse, but I don't know. Do you suggest to close this post and write in the forum?

ilg-ul, 2016-11-23:
If you suspect a bug in the F0 template, please provide more details and try to diagnose the problem.

I suggest you place a breakpoint at main() and see if it reaches this point. If not, place a breakpoint at _start, and step through the initialisation code until it crashes.

deejayfrankcesco, 2016-11-23:
Ok I do this. Anyway the main() is not reached

deejayfrankcesco, 2016-11-23:
I set breakpoints at ./system/src/cortexm/exception_handlers.c -> Reset_Handler (void) and ./system/src/newlib/_startup.c -> _start(void) but It doesn't reach both. It's strange!

ilg-ul, 2016-11-23:
check the ram size in the linker scripts. does it match your device?

deejayfrankcesco, 2016-11-23:
You are great Liviu, FLASH and RAM was wrong! My device has 16K 4K. The template default was 64K 8K. Many Many Thanks

ilg-ul, 2016-11-23:
the flash size is not that important in this context, but if you set the stack outside the existing ram, the first push in the Reset_Handler will crash the device.

as a general remark, I'm always puzzled how fast people click the ok button in the wizard, without checking the actual values in the fields, as the wizard can read their minds and always set the correct values.

and I'm also puzzled how easy people suspect the templates, successfully used in hundreds of projects, instead of their own configurations.

closed 2016-11-24 by ilg-ul