eclipse-embed-cdt/eclipse-plugins #150: Peripheral register addresses are ignoring addressOffset field from cluster in SVD files

TimothyVanRuitenbeek, 2016-11-28:
### Description

The `addressOffset` value in `cluster` tags seems to be ignored when calculating register addresses.  Only the peripheral `baseAddress` and the register `addressOffset` are used.  See the [comment on line 21 of SvdClusterDMNode.java](https://github.com/gnuarmeclipse/plug-ins/blob/master/ilg.gnuarmeclipse.debug.gdbjtag/src/ilg/gnuarmeclipse/debug/gdbjtag/datamodel/SvdClusterDMNode.java#L21) for the intended behavior.

### Steps to Reproduce

1. Install plugin & Nordic Semiconductor nRF_DeviceFamilyPack
2. Set Project's device to `nRF52832_xxAA`
3. Debug the project and open the `PWM0` registers
4. Check the register address values for `SEQ` and `PSEL`

**Expected behavior:** 

`SEQ` should start at address 0x4001C520.
`SEQ->SEQ[0]` should start at address 0x4001C520.
`SEQ->SEQ[1]` should start at address 0x4001C540.
`PSEL` should start at address 0x4001C560.
`PSEL->OUT[0]` should start at address 0x4001C560.
`PSEL->OUT[1]` should start at address 0x4001C564.
`PSEL->OUT[2]` should start at address 0x4001C568.
`PSEL->OUT[3]` should start at address 0x4001C56C.

**Actual behavior:**

`SEQ` address is shown as 0x4001C000 (equal to the baseAddress).
`SEQ->SEQ[0]` starts at address 0x4001C000.
`SEQ->SEQ[1]` starts at address 0x4001C020.
`PSEL` address is shown as 0x4001C000 (equal to the baseAddress).
`PSEL->OUT[0]` starts at address 0x4001C000.
`PSEL->OUT[1]` starts at address 0x4001C004.
`PSEL->OUT[2]` starts at address 0x4001C008.
`PSEL->OUT[3]` starts at address 0x4001C00C.

Modifying the `addressOffset` value for the `OUT[]` register within the `PSEL` cluster from 0x000 to 0x560 (on line 27165 below) updates PSEL->OUT[] to show the correct addresses (and values), but `PSEL` still shows as address 0x4001C000.

Excerpt from `nrf52.svd`:

```
27155         <cluster>
27156           <name>PSEL</name>
27157           <description>Unspecified</description>
27158           <headerStructName>PWM_PSEL</headerStructName>
27159           <addressOffset>0x560</addressOffset>
27160           <register>
27161             <dim>4</dim>
27162             <dimIncrement>4</dimIncrement>
27163             <name>OUT[%s]</name>
27164             <description>Description collection[0]:  Output pin select for PWM channel 0</description>
27165             <addressOffset>0x000</addressOffset>
27166             <access>read-write</access>
27167             <resetValue>0xFFFFFFFF</resetValue>
```

### Versions

* GNU ARM C/C++ Debugging DGB	3.1.1.201606210758	ilg.gnuarmeclipse.debug.gdbjtag
* eclipse.buildId=4.6.0.I20160606-1100
* jdk1.8.0_102
* OS X 10.11.6
* arm-none-eabi-gcc (GNU Tools for ARM Embedded Processors) 5.4.1 20160609 (release) [ARM/embedded-5-branch revision 237715]

ilg-ul, 2016-11-28:
thank you for the detailed bug report; I'll investigate and fix, but it'll take some time (a few weeks?), right now I'm working on a new version of GU ARM Eclipse QEMU.

ilg-ul, 2017-04-24:
the code to extract the `<addressOffset>` from clusters was indeed missing. after adding it I discovered some problems with the recursive processing of offsets, and it took some head scratching to figure out a solution.

I guess it is ok now, versions later than 3.4.1.201704241323 should include the fix.

closed 2017-04-26 by ilg-ul