eclipse-embed-cdt/eclipse-plugins #151: Eclipse Neon.2 GNU ARM install failed

mgiaco, 2016-11-29:
### Description

Installing Software has encountered a problem. I tried the marketplace way, and I also tried the normal update site.

`An error occurred while collecting items to be installed
session context was:(profile=epp.package.cpp, phase=org.eclipse.equinox.internal.p2.engine.phases.Collect, operand=, action=).
Unable to read repository at http://gnuarmeclipse.sourceforge.net/updates/plugins/ilg.gnuarmeclipse.debug.gdbjtag.openocd_4.1.1.201606210758.jar.
sun.security.validator.ValidatorException: PKIX path building failed: sun.security.provider.certpath.SunCertPathBuilderException: unable to find valid certification path to requested target
Unable to read repository at http://gnuarmeclipse.sourceforge.net/updates/plugins/ilg.gnuarmeclipse.debug.packs_1.1.1.201606210758.jar.
sun.security.validator.ValidatorException: PKIX path building failed: sun.security.provider.certpath.SunCertPathBuilderException: unable to find valid certification path to requested target
Unable to read repository at http://gnuarmeclipse.sourceforge.net/updates/features/ilg.gnuarmeclipse.managedbuild.cross_2.2.1.201606210758.jar.
sun.security.validator.ValidatorException: PKIX path building failed: sun.security.provider.certpath.SunCertPathBuilderException: unable to find valid certification path to requested target
Unable to read repository at http://gnuarmeclipse.sourceforge.net/updates/plugins/ilg.gnuarmeclipse.managedbuild.packs_1.3.1.201606210758.jar.
sun.security.validator.ValidatorException: PKIX path building failed: sun.security.provider.certpath.SunCertPathBuilderException: unable to find valid certification path to requested target
Unable to read repository at http://gnuarmeclipse.sourceforge.net/updates/plugins/ilg.gnuarmeclipse.packs_1.2.3.201606210758.jar.
sun.security.validator.ValidatorException: PKIX path building failed: sun.security.provider.certpath.SunCertPathBuilderException: unable to find valid certification path to requested target
Unable to read repository at http://gnuarmeclipse.sourceforge.net/updates/features/ilg.gnuarmeclipse.packs_2.2.2.201606210758.jar.
sun.security.validator.ValidatorException: PKIX path building failed: sun.security.provider.certpath.SunCertPathBuilderException: unable to find valid certification path to requested target
Unable to read repository at http://gnuarmeclipse.sourceforge.net/updates/plugins/ilg.gnuarmeclipse.packs.ui_1.2.1.201606210758.jar.
sun.security.validator.ValidatorException: PKIX path building failed: sun.security.provider.certpath.SunCertPathBuilderException: unable to find valid certification path to requested target
Unable to read repository at http://gnuarmeclipse.sourceforge.net/updates/plugins/ilg.gnuarmeclipse.templates.core_2.5.4.201606210758.jar.
sun.security.validator.ValidatorException: PKIX path building failed: sun.security.provider.certpath.SunCertPathBuilderException: unable to find valid certification path to requested target`

### Steps to Reproduce

1. Install a New Eclipse CDT Neon.2
2. Install the plugins
3. I tried it on 2 different machines

* [plug-in version: latest]
* [Eclipse version: latest Neon.2 CDT version]
* [Java version: 64Bit 1.8.0_92]
* [operating system: win7_64]

ilg-ul, 2016-11-29:
do you use any antivirus?

mgiaco, 2016-11-29:
Yes but I have not rights to disable that at my work pc.

mgiaco, 2016-11-29:
Okay but a colleague of mine also tried it on a different machine he gets the same error - i will ask him if he can try it without antivirus.

ilg-ul, 2016-11-29:
the problem has nothing to do with the plug-ins, the message tells that Eclipse cannot connect to SourceForge, due to some security issues. you either have a firewall or a some antivirus blocking some ports/addresses.

you can always instal the plug-ins offline, download the archive and install from it.

postspectacular, 2016-11-29:
@mgiaco try defining this global environment variable:

`export _JAVA_OPTIONS="-Djava.net.preferIPv4Stack=true -Dhttp.nonProxyHosts=\"*\""`

(I had a similar problem w/ Java + SSL on our corporate firewall/proxy config - setting this solved it)

Also see: http://stackoverflow.com/questions/28327620/difference-between-java-options-java-tool-options-and-java-opts#30305597

ilg-ul, 2016-11-29:
or contact the company network admin to fix his abusively restrictive configurations.

mgiaco, 2016-11-30:
Hello,

@ilg-ul  I tried it on a laptop with no virus and firewall active so there must be an other problem.

@postspectacular - thanks i will try this later, can I add the options in the eclipse.ini file?

mgiaco, 2016-11-30:
So I tied it but then I got these errors:

`An error occurred while collecting items to be installed
session context was:(profile=epp.package.java, phase=org.eclipse.equinox.internal.p2.engine.phases.Collect, operand=, action=).
No repository found containing: osgi.bundle,ilg.gnuarmeclipse.core,3.1.1.201606210758
No repository found containing: osgi.bundle,ilg.gnuarmeclipse.debug.core,1.2.1.201606210758
No repository found containing: org.eclipse.update.feature,ilg.gnuarmeclipse.debug.gdbjtag.jlink,4.1.1.201606210758
No repository found containing: org.eclipse.update.feature,ilg.gnuarmeclipse.debug.gdbjtag.openocd,4.1.1.201606210758
No repository found containing: osgi.bundle,ilg.gnuarmeclipse.debug.gdbjtag.qemu,3.1.1.201606210758
No repository found containing: org.eclipse.update.feature,ilg.gnuarmeclipse.debug.gdbjtag.qemu,3.1.1.201606210758
No repository found containing: osgi.bundle,ilg.gnuarmeclipse.debug.gdbjtag.restart,1.3.1.201606210758
No repository found containing: osgi.bundle,ilg.gnuarmeclipse.debug.packs,1.1.1.201606210758
No repository found containing: osgi.bundle,ilg.gnuarmeclipse.packs,1.2.3.201606210758
No repository found containing: osgi.bundle,ilg.gnuarmeclipse.packs.core,1.1.3.201606210758
No repository found containing: osgi.bundle,ilg.gnuarmeclipse.templates.core,2.5.4.201606210758`

That's really strange I installed this plugin several times in the last years and I had no problems till now.
So I installed it by hand - that works.

I hope I will find the problem later.

thx

mgiaco, 2016-11-30:
I also tried different plugins and I had no issue. So maybe it's a sourceforge problem.

diabolo38, 2016-11-30:
for information hoping it can help 

Some company firewall /proxy requires the eclipse connection setting or proxy to be set or it will never work 
It's my case specilay for  "https" that goes true zscaler proxy  no way to pass it in "native" configuration (what is ok for http)
But i can use a dedicate "application specific proxy"  , get in touch with network admin.

mgiaco, 2016-11-30:
Yes I really know that problem but I tried it over a hotspot with a mobile phone so that should be fine and I get the same error

mgiaco, 2016-11-30:
Oh sorry close the issue i found my problem i had a forefront client active my fault!

closed 2016-11-30 by mgiaco

ilg-ul, 2016-11-30:
and, despite everything, you still love windows...

mgiaco, 2016-11-30:
linux is not possible in our company :-( but i hope some day the will change that

Arsal777, 2016-12-15:
hey @mgiaco and @ilg-ul i have kind of same problem i.e. i can't install the plugins from:  "http://gnuarmeclipse.sourceforge.net/updates" and i am getting following errors that i have placed at the end of my text. I have also tried it offline and online both ways, installed it and uninstalled it many times, did restarts of eclipse and my pc many times still i am getting same errors. I tried on other computers of the same network and it works but at my system its not working and my thesis work is stuck at this point. I have also disabled my antivirus and firewall still its not working. Dear @mgiaco can you please share what you did at your pc in steps so that i can try that at my end.

"An error occurred while collecting items to be installed
session context was:(profile=C__Users_arsalan_eclipse_cpp-neon_eclipse, phase=org.eclipse.equinox.internal.p2.engine.phases.Collect, operand=, action=).
Unable to read repository at https://sourceforge.net/projects/gnuarmeclipse/files/Eclipse/updates/plugins/ilg.gnuarmeclipse.codered_1.1.1.201606210758.jar.
sun.security.validator.ValidatorException: PKIX path building failed: sun.security.provider.certpath.SunCertPathBuilderException: unable to find valid certification path to requested target
Unable to read repository at https://sourceforge.net/projects/gnuarmeclipse/files/Eclipse/updates/features/ilg.gnuarmeclipse.codered_1.1.1.201606210758.jar.
sun.security.validator.ValidatorException: PKIX path building failed: sun.security.provider.certpath.SunCertPathBuilderException: unable to find valid certification path to requested target
Unable to read repository at https://sourceforge.net/projects/gnuarmeclipse/files/Eclipse/updates/plugins/ilg.gnuarmeclipse.core_3.1.1.201606210758.jar.
sun.security.validator.ValidatorException: PKIX path building failed: sun.security.provider.certpath.SunCertPathBuilderException: unable to find valid certification path to requested target
Unable to read repository at https://sourceforge.net/projects/gnuarmeclipse/files/Eclipse/updates/plugins/ilg.gnuarmeclipse.debug.gdbjtag.jlink_4.1.1.201606210758.jar.
sun.security.validator.ValidatorException: PKIX path building failed: sun.security.provider.certpath.SunCertPathBuilderException: unable to find valid certification path to requested target
Unable to read repository at https://sourceforge.net/projects/gnuarmeclipse/files/Eclipse/updates/plugins/ilg.gnuarmeclipse.debug.gdbjtag.openocd_4.1.1.201606210758.jar.
sun.security.validator.ValidatorException: PKIX path building failed: sun.security.provider.certpath.SunCertPathBuilderException: unable to find valid certification path to requested target
Unable to read repository at https://sourceforge.net/projects/gnuarmeclipse/files/Eclipse/updates/features/ilg.gnuarmeclipse.debug.gdbjtag.openocd_4.1.1.201606210758.jar.
sun.security.validator.ValidatorException: PKIX path building failed: sun.security.provider.certpath.SunCertPathBuilderException: unable to find valid certification path to requested target
Unable to read repository at https://sourceforge.net/projects/gnuarmeclipse/files/Eclipse/updates/plugins/ilg.gnuarmeclipse.debug.gdbjtag.pyocd_1.1.1.201606210758.jar.
sun.security.validator.ValidatorException: PKIX path building failed: sun.security.provider.certpath.SunCertPathBuilderException: unable to find valid certification path to requested target
Unable to read repository at https://sourceforge.net/projects/gnuarmeclipse/files/Eclipse/updates/features/ilg.gnuarmeclipse.debug.gdbjtag.pyocd_1.1.1.201606210758.jar.
sun.security.validator.ValidatorException: PKIX path building failed: sun.security.provider.certpath.SunCertPathBuilderException: unable to find valid certification path to requested target
Unable to read repository at https://sourceforge.net/projects/gnuarmeclipse/files/Eclipse/updates/plugins/ilg.gnuarmeclipse.debug.gdbjtag.qemu_3.1.1.201606210758.jar.
sun.security.validator.ValidatorException: PKIX path building failed: sun.security.provider.certpath.SunCertPathBuilderException: unable to find valid certification path to requested target
Unable to read repository at https://sourceforge.net/projects/gnuarmeclipse/files/Eclipse/updates/features/ilg.gnuarmeclipse.debug.gdbjtag.qemu_3.1.1.201606210758.jar.
sun.security.validator.ValidatorException: PKIX path building failed: sun.security.provider.certpath.SunCertPathBuilderException: unable to find valid certification path to requested target
Unable to read repository at https://sourceforge.net/projects/gnuarmeclipse/files/Eclipse/updates/plugins/ilg.gnuarmeclipse.debug.gdbjtag.restart_1.3.1.201606210758.jar.
sun.security.validator.ValidatorException: PKIX path building failed: sun.security.provider.certpath.SunCertPathBuilderException: unable to find valid certification path to requested target
Unable to read repository at https://sourceforge.net/projects/gnuarmeclipse/files/Eclipse/updates/plugins/ilg.gnuarmeclipse.debug.packs_1.1.1.201606210758.jar.
sun.security.validator.ValidatorException: PKIX path building failed: sun.security.provider.certpath.SunCertPathBuilderException: unable to find valid certification path to requested target
Unable to read repository at https://sourceforge.net/projects/gnuarmeclipse/files/Eclipse/updates/plugins/ilg.gnuarmeclipse.doc.user_1.1.1.201606210758.jar.
[... cut ...]
sun.security.validator.ValidatorException: PKIX path building failed: sun.security.provider.certpath.SunCertPathBuilderException: unable to find valid certification path to requested target
Unable to read repository at https://sourceforge.net/projects/gnuarmeclipse/files/Eclipse/updates/features/ilg.gnuarmeclipse.templates.stm_2.5.1.201606210758.jar.
sun.security.validator.ValidatorException: PKIX path building failed: sun.security.provider.certpath.SunCertPathBuilderException: unable to find valid certification path to requested target"

[image]

ilg-ul, 2016-12-15:
http://gnuarmeclipse.github.io/blog/2016/12/02/plugins-install-issue/

mgiaco, 2016-12-15:
Many thanks i will try it later

ilg-ul, 2016-12-15:
just update to latest JDK.

Arsal777, 2016-12-16:
Yes, i tried yesterday and it worked out well. bundle of thanks for your help and i really appreciate your effort.

ilg-ul, 2016-12-16:
you're welcome

jantje, 2016-12-20:
First of all thanks for the plugin?
I'm jantje from the Arduino eclipse plugin named Sloeber.
We are trying to add part of your plugin to the Arduino Eclipse plugin Named Sloeber to enable debugging on the zero. see https://github.com/jantje/arduino-eclipse-plugin/issues/603 for what we are trying to do.

I'm sorry to hit you again with this issue but my jenkins build server fails to access sourceforge as well.
Even though the travis build doesn't fail (and @wimjongman reports success) travis does report similar issues https://travis-ci.org/jantje/arduino-eclipse-plugin/builds/185471974#L247
Here is the build output of jenkins
```
Started by user anonymous
Building in workspace /share/homes/admin/.jenkins/jobs/arduino eclipse plugin build/workspace
Cloning the remote Git repository
Cloning repository https://jantje@github.com/jantje/arduino-eclipse-plugin.git
 > /Apps/bin/git init /share/homes/admin/.jenkins/jobs/arduino eclipse plugin build/workspace # timeout=10
Fetching upstream changes from https://jantje@github.com/jantje/arduino-eclipse-plugin.git
 > /Apps/bin/git --version # timeout=10
 > /Apps/bin/git -c core.askpass=true fetch --tags --progress https://jantje@github.com/jantje/arduino-eclipse-plugin.git +refs/heads/*:refs/remotes/origin/*
 > /Apps/bin/git config remote.origin.url https://jantje@github.com/jantje/arduino-eclipse-plugin.git # timeout=10
 > /Apps/bin/git config --add remote.origin.fetch +refs/heads/*:refs/remotes/origin/* # timeout=10
 > /Apps/bin/git config remote.origin.url https://jantje@github.com/jantje/arduino-eclipse-plugin.git # timeout=10
Fetching upstream changes from https://jantje@github.com/jantje/arduino-eclipse-plugin.git
 > /Apps/bin/git -c core.askpass=true fetch --tags --progress https://jantje@github.com/jantje/arduino-eclipse-plugin.git +refs/heads/*:refs/remotes/origin/*
 > /Apps/bin/git rev-parse refs/remotes/origin/master^{commit} # timeout=10
 > /Apps/bin/git rev-parse refs/remotes/origin/origin/master^{commit} # timeout=10
Checking out Revision 6beee4ee36b80fa5bb05c41e06374bb772dc5f1e (refs/remotes/origin/master)
 > /Apps/bin/git config core.sparsecheckout # timeout=10
 > /Apps/bin/git checkout -f 6beee4ee36b80fa5bb05c41e06374bb772dc5f1e
 > /Apps/bin/git rev-list 525b3454eb5ec4226bd900735ccca5165c778087 # timeout=10
Run condition [Execute Shell] enabling prebuild for step [[Send build artifacts over FTP]]
[workspace] $ /bin/ash /share/CACHEDEV1_DATA/.qpkg/Tomcat/tomcat/temp/hudson872419624714120437.sh
rm: cannot remove `*.tar.gz': No such file or directory
Parsing POMs
Modules changed, recalculating dependency graph
Established TCP socket on 54425
[workspace] $ /share/homes/admin/.jenkins/tools/hudson.model.JDK/java_8/bin/java -cp /share/homes/admin/.jenkins/plugins/maven-plugin/WEB-INF/lib/maven32-agent-1.7.jar:/share/homes/admin/.jenkins/tools/hudson.tasks.Maven_MavenInstallation/maven/boot/plexus-classworlds-2.5.1.jar:/share/homes/admin/.jenkins/tools/hudson.tasks.Maven_MavenInstallation/maven/conf/logging jenkins.maven3.agent.Maven32Main /share/homes/admin/.jenkins/tools/hudson.tasks.Maven_MavenInstallation/maven /share/CACHEDEV1_DATA/.qpkg/Tomcat/tomcat/webapps/jenkins/WEB-INF/lib/remoting-2.53.2.jar /share/homes/admin/.jenkins/plugins/maven-plugin/WEB-INF/lib/maven32-interceptor-1.7.jar /share/homes/admin/.jenkins/plugins/maven-plugin/WEB-INF/lib/maven3-interceptor-commons-1.7.jar 54425
<===[JENKINS REMOTING CAPACITY]===>channel started
Executing Maven:  -B -f /share/homes/admin/.jenkins/jobs/arduino eclipse plugin build/workspace/pom.xml clean verify -Pall,neon
[INFO] Scanning for projects...

[INFO] Computing target platform for MavenProject: com.github.brodykenrick.arduino-eclipse-plugin:io.sloeber.feature:4.0.0-SNAPSHOT @ /share/homes/admin/.jenkins/jobs/arduino eclipse plugin build/workspace/io.sloeber.feature/pom.xml

[INFO] Fetching p2.index from http://download.eclipse.org/eclipse/updates/4.6/ (0B at 0B/s)
[INFO] Adding repository http://download.eclipse.org/eclipse/updates/4.6

[INFO] Fetching p2.index from http://download.eclipse.org/eclipse/updates/4.6/R-4.6-201606061100/ (0B at 0B/s)

[INFO] Fetching p2.index from http://download.eclipse.org/eclipse/updates/4.6/R-4.6.1-201609071200/ (0B at 0B/s)

[INFO] Fetching p2.index from http://download.eclipse.org/releases/neon/ (0B at 0B/s)
[INFO] Adding repository http://download.eclipse.org/releases/neon

[INFO] Fetching p2.index from http://download.eclipse.org/technology/epp/packages/neon/ (0B at 0B/s)

[INFO] Fetching p2.index from http://download.eclipse.org/releases/neon/201606221000/ (0B at 0B/s)

[INFO] Fetching p2.index from http://download.eclipse.org/releases/neon/201609281000/ (0B at 0B/s)

[INFO] Fetching p2.index from http://download.eclipse.org/releases/neon/201610111000/ (0B at 0B/s)

[INFO] Adding repository http://download.eclipse.org/nebula/snapshot

[WARNING] Invalid cookie header: "Set-Cookie: VISITOR=58593fe0cc0a456dcc5d3bd8; expires="Fri, 18-Dec-2026 14:27:44 GMT"; httponly; Max-Age=315360000; Path=/". Unable to parse expires attribute: "Fri, 18-Dec-2026 14:27:44 GMT"
[WARNING] Invalid cookie header: "Set-Cookie: VISITOR=58593fe0cc0a456dcc5d3bd8; expires="Fri, 18-Dec-2026 14:27:45 GMT"; httponly; Max-Age=315360000; Path=/". Unable to parse expires attribute: "Fri, 18-Dec-2026 14:27:45 GMT"

[... cut ...]
	at hudson.model.Executor.run(Executor.java:410)
Caused by: com.sun.mail.smtp.SMTPAddressFailedException: 554 5.7.1 Recipient address rejected: Username jan@baeyens.it and sender nobody@nowhere doesnt match

	at com.sun.mail.smtp.SMTPTransport.rcptTo(SMTPTransport.java:1733)
	... 10 more
Finished: FAILURE
```

FYI:
Sloeber runs on java8 and the product is bundled with eclipse neon.
Jenkins runs on a linux box here at my home without any special setup (be it security/firewall....)

Do you have a clue what is going on?

Thanks
Jantje

ilg-ul, 2016-12-20:
no idea, except using the latest JDK.

jantje, 2016-12-20:
>no idea, except using the latest JDK.
I thought both travis and jenkins used the latest sdk. Why do you think they are not?

ilg-ul, 2016-12-21:
add `java -version` in a script and you'll know for sure.

wimjongman, 2016-12-21:
I have upgraded to the latest SDK at the time of writing (8 u112) and while I still get cookie errors the build succeeds 9 out of 10 times.

I think sourecforge is the problem. Plenty of mentions of this on the net but I cannot find an immediate cause or solution.

jantje, 2016-12-21:
@wimjongman 
I agree with your suspicion towards sourceforge.
 
@ilg-ul 
Would it( be acceptable to you that I build your project on my build server and push a update site to sloeber.io/gnuarmeclipse/nightly?

ilg-ul, 2016-12-21:
> push a update site to sloeber.io/gnuarmeclipse/nightly

ok.

wimjongman, 2016-12-21:
Why build it? Just download it from sourceforge.

shihrer, 2016-12-23:
I've been getting an error trying to load the plugins from gnuarmeclipse.sourceforge.net/updates.  It simply telles me "No software site found at http://gnuarmeclipse.sourceforge.net/updates."  Is this a related issue?

ilg-ul, 2016-12-23:
@shihrer, did you install the latest JDK, as suggested?

jantje, 2016-12-23:
As to Wim his proposal I created http://sloeber.io/gnuarmeclipse/stable which is a copy of the content of  ilg.gnuarmeclipse.repository-3.1.1-201606210758.zip.
I agree with Wim that for Sloeber the stable is the way to go so a manual update process is fine.

@ilg-ul 
If you want I can setup a nightly based on my build engine as well.

PS sorry for the delay but I must have missed the responses

ilg-ul, 2016-12-23:
> a nightly based on my build engine as well.

you can do this, but for now there are not many changes, so it would be a waste. I suggest you follow the git, and when you think appropriate, you can run a new build. or we can configure a hook and you run the build on it.

fyi, I plan some repairs on the plug-ins after I fix qemu, probably in January.

jantje, 2016-12-23:
nightly has been added to the build.
The update site will only change when there has been a change in the repository and the build was successful.

shihrer, 2016-12-23:
@ilg-ul Hey thanks for the response.  I'm using OpenJDK 1.8.0_111.  It works fine if I try to install through the latest version of Eclipse, but I'm running into issues with Kinetis Design Studio which is based on Eclipse 4.4.2.  I was able to get it installed through the update site a while back, so it at least worked at one point.  Trying to get the dev environment setup on another system and now the update site isn't working.  I was able to install using the downloaded archive, though, which is fine for my case.

jantje, 2016-12-23:
To summarize there are 2 new update sites
http://sloeber.io/gnuarmeclipse/stable 
contains 3.1.1
http://sloeber.io/gnuarmeclipse/nightly
contains the master build and is build around 1 CET (uploads only when there are changes)
If the problems are caused by sourgeforge I think this issue can be closed due to "a workaround"

shihrer, 2016-12-23:
@jantje Works.  Thanks.