eclipse-embed-cdt/eclipse-plugins #152: My Workshop at Estácio do Ceará - III Siconect

carlosdelfino, 2016-11-29:
The Workshop was very good, I had some problems, but because I had to look for alternatives to achieve in 2 hours to focus on the theme "Eclipse, QEMU and Cortex-M", without installation details, but without taking away the pleasure of some steps Installation.

I wanted to give you some feedback about the process.

I'm writing everything down at http://carlosdelfino.eti.br/III-SiConect.

I did all the installations in a VM, but I also left to the discretion of the institution the individual installation, which was difficult for the young people responsible for the infra university center, who did not dominate the tool do not know because, everything is very easy without a doubt.

But I realized that downloading CMSIS templates is a long process, and the most costly in time, it would be interesting to leave a hook to search for solutions related to this process to become fast.

To solve such a problem, I left a copy of the "Packages" directory on each machine so that it could be used, would there be a more efficient way for offline installation?

I noticed that even after the package was installed by copying only the folder, and updating, sometimes it was listed correctly in its eclipse perspective.

But other times when clicking the update button all packages disappear, and had to download again online, did not accept the package.

It would be interesting to have a way to download the templates only from the modules and microcontrollers used in Blink's example.

I leave the suggestion as a hook for a future improvement, so I mastered such an implementation I can verify this.

Thanks for your support and codes.
### Prerequisites

In case you encounter problems while using the plug-ins, please consider the following:

* [ ] read the documentation pages carefully
* [ ] check the [Known issues](http://gnuarmeclipse.github.io/support/known-issues/)
* [ ] check the [FAQ](http://gnuarmeclipse.github.io/support/faq/)
* [ ] check the [GNU ARM Eclipse forum](http://gnuarmeclipse.github.io/support/forum/) for similar problems
* [ ] check the GitHub Issues [tracker](https://github.com/gnuarmeclipse/plug-ins/issues/)

If you still could not find a solution, if you have interesting use cases, if you have custom 
configurations, and generally if you have any experience that you want to share with others, 
use the [GNU ARM Eclipse forum](http://gnuarmeclipse.github.io/support/forum/).

If you are convinced that you identified a bug (if you have doubts, use the forum), or you have a 
pertinent suggestion how to enhance the GNU ARM Eclipse plug-ins, continue and register
an issue.

### Description

[Description of the bug or feature]

### Steps to Reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expected to happen]

**Actual behavior:** [What actually happened]

### Versions

* [plug-in version]
* [Eclipse version]
* [Java version]
* [operating system]
* [toolchain version]

Please understand that without being able to reproduce the bug we cannot identify your problem.

---

Note: Remove unnecessary text after reading, before entering your ticket.

ilg-ul, 2016-11-29:
> downloading CMSIS templates is a long process

you mean CMSIS packs? yes, this is a design issue, they want the vendors to pack everything related to a device family in a single archive, and this results in huge files, that take a long time to download.

and also the CMSIS packs do not handle packs dependencies, in order to be able to split large packages in small pieces and use only what is needed.

I already addressed these issues, the xPacks are much smaller, and a dependency mechanism will automatically bring all needed small packages.

> clicking the update button all packages disappear

ouch! the Packs Manager had a bug, that was fixed, but I did not make the public release yet; the beta is available in updates-test, you should have used it.

> It would be interesting to have a way to download the templates only from the modules and microcontrollers used in Blink's example.

if you remember, the µOS++ demo and test projects use a `generate.sh` script, that brings only a few xPacks.

carlosdelfino, 2016-11-29:
yes, now I assimilate better the propose of xPacks.

ilg-ul, 2016-11-29:
> I assimilate better the propose of xPacks

if you have a chance, study the npm mechanism to install JS modules, since the xPacks will share the same servers, and must use a similar mechanism.

closed 2017-02-25 by ilg-ul