eclipse-embed-cdt/eclipse-plugins #153: Template C Project should declare raise() and kill() as weak

ivlz, 2016-12-04:
If one includes <signal.h> and uses the raise() function, then the function raise() defined in <project>/system/src/newlib/_syscalls.c will cause a redefinition error.

  // These functions are defined here to avoid linker errors in freestanding
  // applications. They might be called in some error cases from library
  // code.
  //
  // If you detect other functions to be needed, just let us know
  // and we'll add them.
  int
  raise(int sig __attribute__((unused)))
  {
    errno = ENOSYS;
    return -1;
  }

I fixed the problem by declaring raise() as weak
  __attribute__((weak)) int
  raise(int sig __attribute__((unused)))
  {
    errno = ENOSYS;
   return -1;
 }

In other situations it may be necessary to do the same for kill()
  int
  kill(pid_t pid, int sig);

  __attribute__((weak)) int
  kill(pid_t pid __attribute__((unused)), int sig __attribute__((unused)))
  {
    errno = ENOSYS;
    return -1;
  }

Would be interested to know your thoughts on this, or a better solution to my current workaround.

Thanks.

ilg-ul, 2016-12-04:
are you using newlib or newlib nano?

personally I'm not a big fan of newlib, I think the implementation is not very fortunate in many ways, but currently we have to live with it, and probably use many workarounds.

future versions of the templates will use code from the latest version of [µOS++](http://micro-os-plus.github.io), where I already reimplemented more thoroughly the syscalls.

ivlz, 2016-12-08:
Thanks.

I a believe I am using newlib nano. I created a template project and then pulled in the STM32_NUCLEO sources into the project. I believe that the relevant linker option is  --specs=nano.specs which I have set.

ilg-ul, 2017-04-24:
fixed since 3.4.1.201704240806

closed 2017-04-26 by ilg-ul