eclipse-embed-cdt/eclipse-plugins #155: nRF_DeviceFamilyPack does not update to v8.11.0

bjornspockeli, 2016-12-13:
### Description

Nordic Semiconductor has released a new SDK version, SDK v12.x.0, that requires nRF_DeviceFamilyPack 8.11.0.  The lastest version of the nRF_DeviceFamily_Pack available from the Packs Manager is v8.9.0 and updating the package definitions from all repositories does not add v8.11.0 to the list. 

### Steps to Reproduce

1. Install the pre-release plugin ( http://gnuarmeclipse.sourceforge.net/updates-test) 
2. Press "Update the package definitions from all repositories" in C/C++ Packs perspective window

**Expected behavior:** [What you expected to happen]

nrF_DeviceFamily_Pack v8.11.0 is added to the list over available packs in the Packs Manager. 

**Actual behavior:** [What actually happened]

After updating the packs definitions the latest nrF_DeviceFamily_Pack is v8.9.0.

Checking the CMSIS Pack Repository index file (http://www.keil.com/pack/index.idx) I see that the v8.11.0 pack is added, i.e. the following entry is in the list. 

> pdsc url="http://developer.nordicsemi.com/nRF5_SDK/pieces/nRF_DeviceFamilyPack/" name="NordicSemiconductor.nRF_DeviceFamilyPack.pdsc" version="8.11.1" 

### Versions

* [plug-in version]:

  GNU ARM C/C++ Packs (Experimental)	2.2.2.201611031134

* [Eclipse version]:

Eclipse IDE for C/C++ Developers
Version: Neon.1a Release (4.6.1)
Build id: 20161007-1200

* [Java version]
JRE v1.80_111

* [operating system]
Windows 7 (64-bit)

* [toolchain version]
GNU ARM Embedded Toolchain v4.9 2015 q2

ilg-ul, 2016-12-13:
please try the pre-release version from updates-test

bjornspockeli, 2016-12-13:
I have installed the pre-release version from http://gnuarmeclipse.sourceforge.net/updates-test.

ilg-ul, 2016-12-13:
ok, then it must be something else; I'll investigate, but it might take some time.

ilg-ul, 2016-12-13:
for the records, could you copy here the packs console output and possibly any exceptions in the Eclipse log (workspace/.metadata/.log) related to the problem?

bjornspockeli, 2016-12-13:
I have attached the console output from the GNU ARM Eclipse Packs console and the Eclipse log.

[GNU_ARM_Eclipse_Packs_Console_output.txt](https://github.com/gnuarmeclipse/plug-ins/files/649025/GNU_ARM_Eclipse_Packs_Console_output.txt)
[Eclipse_log.txt](https://github.com/gnuarmeclipse/plug-ins/files/649034/Eclipse_log.txt)

ilg-ul, 2016-12-13:
hmmm... not much so far...

bjornspockeli, 2016-12-13:
_"hmmm... not much so far..."_ as in: a) the files were attached and you're not getting much from them or b) they were not attached?

ilg-ul, 2016-12-13:
they were attached, but, apart the weird exceptions related to icons, I could not get much from them.

bjornspockeli, 2016-12-13:
The nRF_DeviceFamily_Pack v8.11.1 appears to be processed 

Processing "C:/Users/bjsp/Packages/.cache/NordicSemiconductor.nRF_DeviceFamilyPack.8.11.1.pdsc" for content.xml ...

ilg-ul, 2017-04-24:
I just fixed a problem related to ordering the package versions. versions were ordered lexicographically, not using the numeric values, and 8.11 ended close to 8.1, down the list.

please try the 3.4.1 beta from updates-test

ilg-ul, 2017-04-24:
related to #157

closed 2017-04-26 by ilg-ul