eclipse-embed-cdt/eclipse-plugins #156: Unable to start a debug session after updating to Eclipse Neon.2 (4.6.2)

stf12, 2016-12-22:
### Description

The issue happens after I updated from Eclipse 4.6.1 to Eclipse 4.6.2.
Moreover I have installed the last version of:
- GNU ARM Eclipse plugin
- GNU ARM Eclipse OpenOCD
- GNU ARM Eclipse Build Tool for windows
- GNU ARM Embedded Toolchain 5-2016-q3-update 

When I try to lunch debug session of type "GDB OpenOCD Debugging", the launch sequence is blocked at 99% (Launching: Configuring GDB) and the debugger never starts.

### Steps to Reproduce

1. I use a C++ program for a NUCLEO-F410 board created using the New Project wizard.
2. I create a standard Debug configuration 
3. I launch the debug configuration

**Expected behavior:** [What you expected to happen]
I expect for the debugger to start.

**Actual behavior:** [What actually happened]
The launch sequence is blocked at 99% (Launching: Configuring GDB) and the debugger never starts.

### Versions

* [plug-in version]
3.1.1-201606210758
* [Eclipse version]
4.6.2
* [Java version]
JRE 1.8.0_111
* [operating system]
Windows 7 64 bit
* [toolchain version]
gcc-arm-none-eabi-5_4-2016q3-20160926-win32

ilg-ul, 2016-12-22:
I confirm the problem, but it seems specific to Neon and Windows, on my Mac it is ok.

As a workaround, you can try to use Mars.2.

stf12, 2016-12-25:
Ok. I will wait for the fix. If I can find some more information I will report here.

Thank you.

ilg-ul, 2016-12-25:
I informed the CDT developers. I do not know if there is anything I can do in the plug-ins, or the CDT needs to repaired, and in this case it might take a while.

Menion2k, 2017-01-04:
Problem still present with CDT 9.2 on Neon.2. The problem was also present running CDT 9.0 on Neon.2 as well

ilg-ul, 2017-01-04:
the current recommendation is to stay away from Neon and use Mars.2.

Menion2k, 2017-01-04:
Actually it was working perfectly with Neon.1 and CDT 9.0

ilg-ul, 2017-01-04:
please do not reply by email, but use the web interface, or at least do not quote the original message.

Menion2k, 2017-01-05:
The problem is also present in Eclipse Oxygen + CDT milestone

ilg-ul, 2017-01-05:
> Could you share the bugzilla please?

I did not register a bugzilla yet, I used the cdt-dev list.

johnmoule, 2017-01-05:
Thanks for the info.

@Menion2k 
>  I have just wiped my Neon.2 and reinstalled the above with gnuarmeclipse 3.11 and it works perfectly (Windows Host).

Which version of CDT are you using with that?
Can I infer from this discussion that the problem appears to be not with neon.2 but with cdt9.2.0?

Menion2k, 2017-01-05:
@johnmoule I have written it in my post: CDT 9.1 and no, the problem is of Neon.2 not CDT

mibcat, 2017-01-10:
shit happens - just updated from Mars to Neon yesterday and running into the same trouble ...

ilg-ul, 2017-01-10:
the penalty for not reading the installation notes.

mibcat, 2017-01-10:
you make a good point ! ;-)

ilg-ul, 2017-01-14:
fixed in v3.2.1-201701141320

closed 2017-01-14 by ilg-ul

stf12, 2017-01-16:
Hi Liviu,

thank you for support. I confirm that the issue is solved after updating to the new version :).

But now I have another strange issue that already happened in the past. The New Project Wizard doesn't display the STM32 Template, even if in the Eclipse "Preferences > C/C++ > New C/C++ Project Wizard" I can see them (see the below images).

Btw, I think this is my problem.

Thank you again,
Stefano

[image]

[image]

ilg-ul, 2017-01-16:
does your problem has anything to do with the issue subject?

for general questions, please use the project forum.

stf12, 2017-01-16:
Hi Liviu,

this problem is happening after I updated the GNU ARM plug-in to v3.2.1-201701141320. But after a first analysis, I'm quite sure that is not related with this issue and with the GNU ARM plug-in.

I'm sorry I wasted your time.

ilg-ul, 2017-01-16:
in case further investigations show that it is related, please open a separate ticket.