eclipse-embed-cdt/eclipse-plugins #158: Unable to read repository at http://gnuarmeclipse.sourceforge.net/updates/content.xml.xz.

wiesener, 2016-12-27:
Trying to install the plugin in eclipse neon.2 under mac osx (sierra) brings the following error:

Unable to read repository at http://gnuarmeclipse.sourceforge.net/updates/content.xml.xz.

Either manually or viar marketplace did not work.

Best regards,

C.W.

ilg-ul, 2016-12-27:
duplicate of #151

wiesener, 2016-12-27:
ups sorry, google did not bring it up.
— 
Constantin Wiesener
Technische Universität Berlin,  FG Regelungssysteme (Control Systems Group)
Room: EN 204   -   Phone: +49-30-314-24094   -   Fax: +49-30-314-21137
Post: TU Berlin  -  Sek. EN 11  -  Einsteinufer 17  -  D-10587 Berlin, Germany

ilg-ul, 2016-12-27:
update to latest JDK and it should work.

closed 2016-12-27 by ilg-ul

jantje, 2016-12-27:
I don't want to mix in stuff that is not my business ... but I think "known limitation" covers this issue better than "can't fix"