eclipse-embed-cdt/eclipse-plugins #159: Debug Symbol in Release Build Configuration

pearson, 2016-12-28:
When a new "Hello World ARM Cortex-M C/C++ Project" is created, the OS_USE_TRACE_SEMIHOSTING_DEBUG flag is set for the default **release** build. This is relatively minor, but removing that symbol reduced my release binary size by about 2%. This flag also seems to be set in other prototype applications.

How to reproduce:

1. Create a new project using File -> New -> C Project
2. In the "C Project" dialog, Choose "Hello World ARM Cortex-M C/C++ Project" and "Cross ARM GCC" Toolchain
3. Press Next
4. In the "Target processor settings" dialog, set "Trace output" to "Semihosting DEBUG channel". The other options in this dialog do not change the outcome.
5. Press Next
6. Press Next in the "Folders" dialog
7. In the "Select Configurations" dialog, either:
     - Click the "Advanced settings..." button to open the project properties dialog
     - Click the "Finish" button to create the project, then open project properties dialog via the Window menu
8. In the "Properties for project_name" dialog, choose C/C+ General -> Paths and Symbols
9. Click on the "Symbols" tab
10. Use the "Configuration" pull-down menu to switch the configuration to "Release" to see that the OS_USE_TRACE_SEMIHOSTING_DEBUG flag is set.

Versions:
- Eclipse Neon.1 Linux 64 bit
- GNU ARM C/C++ Cross Compiler	2.2.1.201606210758
- GNU ARM C/C++ Generic Cortex-M Project Template	1.3.1.201606210758
- GNU ARM C/C++ J-Link Debugging	4.1.1.201606210758
- GNU ARM C/C++ OpenOCD Debugging	4.1.1.201606210758
- GNU ARM C/C++ Packs (Experimental)	2.2.2.201606210758
- GNU ARM C/C++ QEMU Debugging	3.1.1.201606210758
- GNU ARM C/C++ STM32Fx Project Templates	2.5.1.201606210758

ilg-ul, 2016-12-28:
ok, thank you, I'll check and probably remove in future releases.

ilg-ul, 2017-04-24:
fixed since 3.4.1.201704241811

actually all `OS_USE_TRACE_*` were removed from release, and `NDEBUG` was added to release.

closed 2017-04-26 by ilg-ul