eclipse-embed-cdt/eclipse-plugins #160: Multicore SVD file parsing

kduca, 2016-12-30:
### Description
No peripherals displayed in Peripherals view while debugging an NXP LPC4350 (multicore: M4 and M0) even though the LPC4300 Series Device pack is installed and associated with the project.

### Steps to Reproduce
1. Make sure LPC4300 Series Device pack is installed (v2.7.0 used).
2. Make sure LPC4350 Device is associated with the M4 code project (Project Properties->C/C++ Build->Settings).
3. Launch M4 debug session and verify it successfully connects to target and stops at main().
4. Select Peripherals view.

**Expected behavior:**
List of peripherals are displayed, or a way to select/expand peripherals in the view.

**Actual behavior:**
The Console view displays `Parsing SVD file "/Users/kd/Documents/GNUARMEclipse/Packages/Keil/LPC4300_DFP/2.7.0"...` but often `Failed to get the peripheral descriptions` displays briefly in red at the bottom status line of the Console view.

The .log file shows:
`!ENTRY ilg.gnuarmeclipse.debug.gdbjtag.pyocd 4 1 2016-12-30 11:53:43.592
!MESSAGE Internal Error
!STACK 0
java.io.FileNotFoundException: /Users/kd/Documents/GNUARMEclipse/Packages/Keil/LPC4300_DFP/2.7.0 (Is a directory)
  at java.io.FileInputStream.open0(Native Method)
  at java.io.FileInputStream.open(FileInputStream.java:195)
  at java.io.FileInputStream.<init>(FileInputStream.java:138)
  at ilg.gnuarmeclipse.core.Xml.parseFile(Xml.java:39)
  at ilg.gnuarmeclipse.debug.gdbjtag.datamodel.SvdUtils.getTree(SvdUtils.java:201)
  at ilg.gnuarmeclipse.debug.gdbjtag.services.PeripheralsService.getPeripherals(PeripheralsService.java:170)
  at ilg.gnuarmeclipse.debug.gdbjtag.viewmodel.peripherals.PeripheralsVMNode.updateElementsInSessionThread(PeripheralsVMNode.java:258)
  at org.eclipse.cdt.dsf.ui.viewmodel.datamodel.AbstractDMVMNode$3.run(AbstractDMVMNode.java:239)
  at java.util.concurrent.Executors$RunnableAdapter.call(Executors.java:511)
  at java.util.concurrent.FutureTask.run(FutureTask.java:266)
  at java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.access$201(ScheduledThreadPoolExecutor.java:180)
  at java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.run(ScheduledThreadPoolExecutor.java:293)
  at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1142)
  at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:617)
  at java.lang.Thread.run(Thread.java:745)`

### Versions
* GNUARMEclipse "Update Test" plugins 20161103
* Neon.2 (4.6.2.M20161124-1400)
* jdk1.8.0_112
* OSX 10.11.6
* Sourcery G++ Lite 2011.03-42 (4.5.2)
* JLink Ultra+

hriekehof, 2017-03-08:
Hi,

i experience the same problem with LPC4300 and imx6 and imx7 packs. Maybe it helps as a hint, If i use the 2.6 Version of the LPC4300 Pack the parsing of the svd file works fine. With the 2.7 Version it stopped working.

- GNU ARM C/C++ Packs (Experimental)	2.2.2.201702251311
- Eclipse Neon2 64-Bit on Windows 10

ilg-ul, 2017-03-08:
I acknowledge the bug, multi-core devices were not considered when writing the plug-ins.

fixing it requires some more efforts; it'll be done, but it might take a while.

hriekehof, 2017-03-08:
Thanks for the quick answer. What i dont understand right now, if multicore devices were not considered why is version 2.6 of the pack working ?

ilg-ul, 2017-03-08:
I guess by pure chance.

hriekehof, 2017-03-13:
Hi, 

regarding this topic again, i had a look into the IMX6_DFP pack from Keil. If i'm not wrong in the SVD folder there is one .svd file per core. Shouldn't it then be possible to parse the xml like for a single core ?

ilg-ul, 2017-03-13:
I don't think the problem is due to parsing multiple files, I guess that selecting the right one confuses the plug-in.

hriekehof, 2017-03-13:
Ah ok i see. But this seems easy to fix :) Sorry for bothering you so hard on this topic but i really would need it :)

ilg-ul, 2017-03-13:
it is on my todo list, but, unless some unexpected contributions that might boost priorities, the priority is not very high.