eclipse-embed-cdt/eclipse-plugins #161: MK22FN128VLH10  in GNU ARM Eclipse

carlosdelfino, 2017-01-04:
Hi @ilg-ul .

I'm studying the use of Freescale's MK22FN128VLH10 in small project, I saw that the GNU ARM Eclipse only has the KL family, is there any special procedure I can follow to use a layout for the MK22 family?

Or should I start with the KL family and make the file adjustments for the MK22?

Grateful.

ilg-ul, 2017-01-05:
for NXP devices probably the easiest way is to use KDS to create the project.

carlosdelfino, 2017-01-05:
I'm going to start studying Kinetics today, I saw that it is possible to install the plugins for eclipse, I want to continue to keep everything in Eclipse.

Thank you.

closed 2017-01-05 by carlosdelfino

carlosdelfino, 2017-01-05:
Is the GNU ARM Eclipse referenced in NXP Kinets is yours?

ilg-ul, 2017-01-05:
yes.