eclipse-embed-cdt/eclipse-plugins #167: Neon.2 (4.6.2) - problem during debug - pause and stop functions don't work

asier70, 2017-01-20:
Problem exists in windows with Neon.2 (4.6.2), GDB 9.2 and all plugings up to date (GnuArmEclipsePlugin, OpenOCD, gcc, stlink). Almost everything works (debug start, trace, breakpoints, resuming after breakpoint) but after clicking on pause or stop microcontroller still works. To exit this not normal situation I must stop OpenOCD (MI error info appears).
Everything works on Mars.2, also project worked on Neon.1 some time ago.

ilg-ul, 2017-01-20:
did you update the plug-ins recently?

asier70, 2017-01-20:
Yes, I did. I have GNU ARM Eclipse plug-ins v3.2.1-201701141320. Some time ago everything worked, but now after do updates in eclipse problem occurs.

ilg-ul, 2017-01-20:
sorry, with the information you provided I cannot reproduce your problem. on my mac, with GCC 5.4, GDB 7.10, latest plug-ins, latest OpenOCD, on my F4DISCOVERY board, I can halt or stop the debug session without problems.

GDB 9.2 seems not correct.

ilg-ul, 2017-01-20:
I also tested on my Win 8 x64 and I can halt or stop the debugger.

asier70, 2017-01-20:
Win 10 x64, GCC 6.2, C/C++ GDB Hardware Debugging	9.2.0.201612061315
Maybe someone else has the same problem.
Thanks for checking and help!

ilg-ul, 2017-01-20:
GCC 6.2 is not yet supported.

asier70, 2017-01-20:
GCC 6.2 2016q4 works on Mars.2 with my project.

ilg-ul, 2017-01-20:
try GCC 5.4 with Neon.2.

asier70, 2017-01-20:
I've just found eclipse bug published by Freddie Chopin, that is the same I described here:
[https://bugs.eclipse.org/bugs/show_bug.cgi?id=509788](url)

ilg-ul, 2017-01-20:
ok, thank you.

wrljet, 2017-01-20:
For what it's worth, things work correctly for me.  I think I have the same setup, GCC 5.4 with Neon.2, latest ARM plugin, *except*, I'm running OpenOCD 0.7.0.

And by the way, a HUGE thank-you, Liviu for all your work.
-Bill

ilg-ul, 2017-01-20:
you're welcome, Bill.

> OpenOCD 0.7.0.

I guess it should also work with 0.10.0

the problem seems to be caused by the combination GDB 7.12, that comes with GCC 6.x, and Neon.2.

asier70, 2017-01-23:
Liviu you were right. Thanks!
Downgrading GCC 6.2 2016q4 (GDB 7.12.0.20161204) to GCC 5.4 2016q3 (GDB 7.10.1.20160923) solves the problem.

Amcolex, 2017-01-23:
I'd just like to add that we had the same problem. Windows 10 / Eclipse Neon.2 /  GCC 6.2 2016q4 / Segger J-Link V612f.

I guess this is a CDT bug though. I'll look into downgrading to GCC 5.4.

And thank you Liviu for all your work!

asier70, 2017-03-24:
Yesterday Eclipse was updated to version Neon.3 (4.6.3). There is no problem in new version with debugging using GCC in version "6.2 2016q4" and newest "6 2017-q1-update".

ilg-ul, 2017-03-24:
great! thank you for testing.

can we close this ticket?

asier70, 2017-03-24:
I've just tested it. I think we can close this ticket.

closed 2017-03-24 by ilg-ul