eclipse-embed-cdt/eclipse-plugins #170: Device name of GDB QEMU Debugging configuration does trigger a change that can be saved

oallaire, 2017-01-25:
### Description

When editing a QEMU Debug Configuration (Debug Configurations -> GDB QEMU Debugging configuration), changing the Device name field of the Debugger tab does enable the apply button.

In order to save the modified Device name, you have to modify another field (say add and remove a character). You will then see the apply button as enabled.

### Steps to Reproduce

1. Modify the Device name
1. Try to click apply

**Expected behavior:** 

The expected behavior would be to have the apply button enabled when you modify the Device name field. 

**Actual behavior:**

The apply button stays disabled (grayed out).

### Versions

* GNU ARM C/C++ QEMU Debugging version 3.1.2.201701141320
* Eclipse Version: Mars.1 Release (4.5.1)
* Java version 1.8.0_77
* Windows 10 x64
* Toolchain version 5.2 2015q4

Thanks!

oallaire, 2017-01-27:
Hi!

I realize now that I could of fixed this bug myself. Do you mind if I take a shot at it and submit a pull request? I'm asking in case you already fixed it.

ilg-ul, 2017-01-27:
you can take a look, but I know what is wrong, there is a callback missing on that widget. 

I planned to fix the problem these days.

ilg-ul, 2017-01-27:
please try 3.3.1.201701271724 from updates-test

oallaire, 2017-01-29:
It works, thanks for the quick fix!

Note that to do the update, I had to follow the instructions here: https://github.com/gnuarmeclipse/plug-ins/issues/171, but it worked after applying these steps.

ilg-ul, 2017-02-25:
fixed since 3.3.1-201702251311

closed 2017-02-25 by ilg-ul