eclipse-embed-cdt/eclipse-plugins #171: handshake_failure when trying to install plugin

monkbroc, 2017-01-27:
Sorry to bother you with yet another plugin install issue.  I read http://gnuarmeclipse.github.io/blog/2016/12/02/plugins-install-issue/ and the closed issues related to plugin install but haven't managed to solve the issue.

I get the "Unable to read repository at http://gnuarmeclipse.sourceforge.net/updates/content.xml." when installing with ProvisionException and handshake_failure.

```
Unable to read repository at http://gnuarmeclipse.sourceforge.net/updates/content.xml.
Received fatal alert: handshake_failure
```
[image]

I tried both on Windows and on Linux with Eclipse Luna, Neon and Neon.2 and with JRE 8u121 and JDK 8u121. I get the error in all cases. I'm doing this from a network without a proxy or firewall, and I don't have an antivirus on Linux and disabled the default Windows Defender. I get the error regardless if I use Eclipse Marketplace or the Install new software menu.

Latest and greatest Eclipse configuration:
[image]

Do you have any other suggestions on what to try?

ilg-ul, 2017-01-27:
it is hard to say. can you install from a local archive? can you install from updates-test?

I don't know what machines you are using, but I'd favour 64-bits JDK and Eclipse.

monkbroc, 2017-01-28:
I'll try 64 bit versions.

Yesterday another developer tried to uninstall and reinstall the plugin on a machine that had the plugin working and got the same error.

Could you please try an install? Maybe SourceForge changed their SSL setup again, maybe due to the OpenSSL security advisory of January 26, 2017.

ilg-ul, 2017-01-28:
you are right, I can no longer install from SourceForge, Eclipse fails with:

```
!SUBENTRY 1 org.eclipse.equinox.p2.transport.ecf 4 1002 2017-01-28 18:48:05.200
!MESSAGE Unable to read repository at http://gnuarmeclipse.sourceforge.net/updates-test/content.xml.
!STACK 0
javax.net.ssl.SSLHandshakeException: Received fatal alert: handshake_failure
	at sun.security.ssl.Alerts.getSSLException(Unknown Source)
	at sun.security.ssl.Alerts.getSSLException(Unknown Source)
	at sun.security.ssl.SSLSocketImpl.recvAlert(Unknown Source)
	at sun.security.ssl.SSLSocketImpl.readRecord(Unknown Source)
	at sun.security.ssl.SSLSocketImpl.performInitialHandshake(Unknown Source)
	at sun.security.ssl.SSLSocketImpl.startHandshake(Unknown Source)
	at sun.security.ssl.SSLSocketImpl.startHandshake(Unknown Source)
	at org.apache.http.conn.ssl.SSLSocketFactory.connectSocket(SSLSocketFactory.java:535)
	at org.apache.http.conn.ssl.SSLSocketFactory.connectSocket(SSLSocketFactory.java:403)
	at org.apache.http.impl.conn.DefaultClientConnectionOperator.openConnection(DefaultClientConnectionOperator.java:177)
	at org.apache.http.impl.conn.ManagedClientConnectionImpl.open(ManagedClientConnectionImpl.java:304)
	at org.apache.http.impl.client.DefaultRequestDirector.tryConnect(DefaultRequestDirector.java:611)
	at org.apache.http.impl.client.DefaultRequestDirector.execute(DefaultRequestDirector.java:446)
	at org.apache.http.impl.client.AbstractHttpClient.doExecute(AbstractHttpClient.java:863)
	at org.apache.http.impl.client.CloseableHttpClient.execute(CloseableHttpClient.java:82)
	at org.eclipse.ecf.provider.filetransfer.httpclient4.HttpClientFileSystemBrowser.runRequest(HttpClientFileSystemBrowser.java:259)
	at org.eclipse.ecf.provider.filetransfer.browse.AbstractFileSystemBrowser$DirectoryJob.run(AbstractFileSystemBrowser.java:69)
	at org.eclipse.core.internal.jobs.Worker.run(Worker.java:54)
```

I'll investigate. in the mean time, download the archive and install it locally.

cnoviello, 2017-01-29:
This stackoverflow answer fixes the issue:

http://stackoverflow.com/a/38264878

ilg-ul, 2017-01-29:
great! 

I'll add a page to the docu site to explain this.

(but for long term, I'm searching for a solution to host the update sites in a different location)

cnoviello, 2017-01-29:
I agree. SF is no longer a good place to store good software ;-)

monkbroc, 2017-01-29:
I got the plugin working with a manual install.

Thank you for your efforts in maintaining this plugin! I'm excited to upgrade from text mode gdb to Eclipse for embedded debugging.

ilg-ul, 2017-01-29:
thank you @monkbroc, good luck with your projects!

ilg-ul, 2017-01-29:
@monkbroc, unless you are in Iran or other difficult location, I suggest you install 'Java Cryptography Extension (JCE) Unlimited Strength Jurisdiction Policy Files 8 Download', as Carmino suggested, for me it solved the problem.

wrljet, 2017-02-21:
'Java Cryptography Extension (JCE) Unlimited Strength Jurisdiction Policy Files 8 Download' worked for me as well.  
In Windows 7 the two jar files went in:  C:\Program Files (x86)\Java\jre1.8.0_121\lib\security

ilg-ul, 2017-02-25:
http://gnuarmeclipse.github.io/blog/2017/01/29/plugins-install-issue/

closed 2017-02-25 by ilg-ul

jrohit54, 2018-04-07:
the issue got resolved bu install the other version of java