eclipse-embed-cdt/eclipse-plugins #172: JLink Debugging. Suspend/Pause not possible in Eclipse Neon.2

Railcrafti, 2017-01-28:
Hi,

I've got a problem when debugging a STM32F407 with the JLink plugin.
Downloading/starting the program and setting breakpoints works. The program stops at the breakpoints.

However, clicking on the "Pause" icon in the debug window has no effect and results in a broken debugging session. 
The JLinkGDBServer's output shows nothing when trying to halt the target.

I also teset this with Eclipse Mars.2. It works properly.

ilg-ul, 2017-01-28:
GDB 7.12 from GCC 6.2 does not work with Neon.2.

Railcrafti, 2017-01-28:
Okay. I have GDB version 7.12. This explains it.
Can you tell me why it doesn't work with Neon.2?

ilg-ul, 2017-01-28:
GDB 7.12 has a new special channel to communicate with the IDE. Neon.2 tried to implement it, but there are some bugs.

on macOS it does not work at all, GDB 7.12 crashes with SIGABORT when accessed via the MI2 interface.

to overcome the first problem, I tried to disable the special console.

please try the version form updates-test and let me know if it works.

on macOS there is nothing to do with GDB 7.12; if you want to use GCC 6.2, rename gdb and make a link to the one from 5.4.

Railcrafti, 2017-01-28:
The testing version also doesn't work.
Now that I'm aware of the cause I'll use an older Eclipse version until the problem is solved

Thank you for your fast answers.

closed 2017-01-28 by Railcrafti