eclipse-embed-cdt/eclipse-plugins #174: Suggestion: Show command line in console

daveloyall, 2017-02-01:
I think that when you invoke Debug, it should some how print the entire command line it will invoke to the console.

ilg-ul, 2017-02-01:
right click on the process name in the tree gives you the full command line.

daveloyall, 2017-02-01:
Doesn't work when qemu crashes immediately.

ilg-ul, 2017-02-24:
fixed since 3.3.1-201702241049

however, printing the command line to console did not help much, since the console output is normally not visible.

the solution I used was to log the line in the Eclipse log file (.metadata/.log); it is not more visible either, but at least it is persistent.

daveloyall, 2017-02-24:
@ilg-ul, thank you.  This will probably help somebody one day!

Cheers,
--Dave

closed 2017-02-25 by ilg-ul