eclipse-embed-cdt/eclipse-plugins #175: Print Size tool/build step - small issue with Hex option

TM1234, 2017-02-03:
I'm using the latest 3.2.1 plugins in Eclipse/CDT Luna SR2 32 bit on Windows 7 64 bit.
I've noticed a small issue with the Cross ARM GNU Print Size tool/build step configuration.
View the properties of a project.
By default Size format = Berkeley and the Hex and Show totals checkboxes are clear.
Select Size format = SysV.
The Hex option becomes greyed out because it's only relevant to Size format = Berkeley.
Click OK to apply/save the changes and close the Properties window.
View the properties of the project again.
Format size = SysV but the Hex option is no not visible at all.
Select Format size = Berkelely and the Hex option remains hidden.
Click OK to apply/save the changes and close the Properties window.
Open the properties again.
Format size = Berkeley and the Hex option is again visible.
I suppose ideally the Hex option should remain visible no matter what Size format is selected but should be disabled/greyed out when Size format = SysV?
Hope this helps.

ilg-ul, 2017-02-03:
thank you, time permitting, I'll try to address this.

TM1234, 2017-02-03:
Thanks for the quick reply Liviu.
As I said it's a minor issue in the greater scheme of things especially given that there is an easy workaround.

ilg-ul, 2017-02-24:
fixed since 3.3.1-201702240920

closed 2017-02-25 by ilg-ul