eclipse-embed-cdt/eclipse-plugins #178: Debug Configuration do not work correctly

Meghabber, 2017-02-23:
Hi,
My conf: Windows 8 64bit, Neon 2.6.4, GNU ARM Eclipse  ttolchain 5.4 2016q3, Build tool .8-201611221915, Qemu plug-in 2.8.0-201612271623-
i have probleme when i launch a debug configuration.  I followed your procedure written in you page but i do not have the "Decision check Box".
here what i had in my console by clicking on Debug button ( Debug configuration box)

Thanks

### Important notice

SourceForce recently updated the security certicates, making Eclipse install/upgrades fail with messages like this:

```
No repository found at http://gnuarmeclipse.sourceforge.net/updates
```

The solution is to upgrade JDK to a version higher than 8u101.

For more details see [GNU ARM Eclipse plug-ins sun.security.validator.ValidatorException](http://gnuarmeclipse.github.io/blog/2016/12/02/plugins-install-issue/).

### Prerequisites

In case you encounter problems while using the plug-ins, please consider the following:

* [ ] read the documentation pages carefully
* [ ] check the [Known issues](http://gnuarmeclipse.github.io/support/known-issues/)
* [ ] check the [FAQ](http://gnuarmeclipse.github.io/support/faq/)
* [ ] check the [GNU ARM Eclipse forum](http://gnuarmeclipse.github.io/support/forum/) for similar problems
* [ ] check the GitHub Issues [tracker](https://github.com/gnuarmeclipse/plug-ins/issues/)

If you still could not find a solution, if you have interesting use cases, if you have custom 
configurations, and generally if you have any experience that you want to share with others, 
use the [GNU ARM Eclipse forum](http://gnuarmeclipse.github.io/support/forum/).

If you are convinced that you identified a bug (if you have doubts, use the forum), or you have a 
pertinent suggestion how to enhance the GNU ARM Eclipse plug-ins, continue and register
an issue.

### Description

[Description of the bug or feature]

### Steps to Reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expected to happen]

**Actual behavior:** [What actually happened]

### Versions

* [plug-in version]
* [Eclipse version]
* [Java version]
* [operating system]
* [toolchain version]

Please understand that without being able to reproduce the bug we cannot identify your problem.

---

Note: Remove unnecessary text after reading, before entering your ticket.

ilg-ul, 2017-02-23:
what exactly is the bug you are reporting? are you sure it is a bug?

> i do not have the "Decision check Box"

what "Decision check Box"?

Meghabber, 2017-02-23:
to honnest, I am not sure it is a bug.
i tried to create a configuration Debug by applying steps written in http://gnuarmeclipse.github.io/debug/qemu/#qemu-plug-in-usage
Steps to Reproduce:
1-new c project was created ( Hello World ARM Cortex-M)
2-Cortex-M3 are selected in " Device Setting" . Note that the pack of Cortex M3 is installed
3-Run ->Debug Configuration: i Followed steps in  http://gnuarmeclipse.github.io/debug/qemu/#qemu-plug-in-usage
*
Expected behavior:  have  a Decision check box (tittltd: Confirm Perspective Switch)
Actual behavior: I did not  have this windows . i had just messages in my console :  
in console:

GNU ARM Eclipse 64-bits QEMU v2.8.0 (qemu-system-gnuarmeclipse.exe).

Supported MCUs:
  STM32F051R8
  STM32F103RB
  STM32F107VC
  STM32F405RG
  STM32F407VG
  STM32F407ZG
  STM32F411RE
  STM32F429ZI

[image]

ilg-ul, 2017-02-23:
> to honnest, I am not sure it is a bug.

then why did you register a bug report, an incomplete one, btw?

isn't the following message clear enough?

> If you are convinced that you identified a bug (if you have doubts, use the forum), or you have a
pertinent suggestion how to enhance the GNU ARM Eclipse plug-ins, continue and register
an issue.

> Expected behavior: have a Decision check box

this is completely non-relevant, it might depend on the Eclipse version (which you did not mention, as requested).

please follow the [bliny tutorial](http://gnuarmeclipse.github.io/tutorials/blinky-arm/) by the book, and you'll get a functional example.

closed 2017-02-23 by ilg-ul