eclipse-embed-cdt/eclipse-plugins #184: Eclipse GNU ARM Plug Plug-in Unable to read repository (Peer not authenticated)

Joohanness, 2017-03-03:
Hello, 

i was trying to add the GNU ARM Plug-In in Eclipse (http://gnuarmeclipse.sourceforge.net/updates) but i always get this log: 

> Unable to read repository at http://gnuarmeclipse.sourceforge.net/updates/content.xml.
Unable to read repository at http://gnuarmeclipse.sourceforge.net/updates/content.xml.
peer not authenticated

I also attached a Picture, which Shows the error. 

I hope someone can help me to solve this Problem. 

Eclipse:
Version: Kepler Service Release 2
Build id: 20140224-0627

Java:
Version 8 Update 121
Build: 1.8.0_121-b13

[image]

ilg-ul, 2017-03-03:
http://gnuarmeclipse.github.io/blog/2017/01/29/plugins-install-issue/

Joohanness, 2017-03-03:
Thank you for your answer, your advice solved my Problem but an other log have occurred. 

> Cannot complete the install because one or more required items could not be found.
  Software being installed: GNU ARM C/C++ Cross Compiler 2.3.1.201702251311 (ilg.gnuarmeclipse.managedbuild.cross.feature.group 2.3.1.201702251311)
  Missing requirement: GNU ARM C/C++ Core 3.3.1.201702251311 (ilg.gnuarmeclipse.core 3.3.1.201702251311) requires 'bundle org.eclipse.cdt 8.6.0' but it could not be found
  Cannot satisfy dependency:
    From: GNU ARM C/C++ Cross Compiler 2.3.1.201702251311 (ilg.gnuarmeclipse.managedbuild.cross 2.3.1.201702251311)
    To: bundle ilg.gnuarmeclipse.core 0.0.0
  Cannot satisfy dependency:
    From: GNU ARM C/C++ Cross Compiler 2.3.1.201702251311 (ilg.gnuarmeclipse.managedbuild.cross.feature.group 2.3.1.201702251311)
    To: ilg.gnuarmeclipse.managedbuild.cross [2.3.1.201702251311]

ilg-ul, 2017-03-03:
http://gnuarmeclipse.github.io/plugins/install/#eclipse--cdt

Joohanness, 2017-03-03:
Thank you your advices helped me to solve all my Problems. Thank you very much.

ilg-ul, 2017-03-03:
you are welcome. I guess we can close this ticket now.

closed 2017-04-23 by ilg-ul

nayefroqaya, 2017-06-05:
I had the same problem but with installing scala 

http://download.scala-ide.org/sdk/lithium/e46/scala212/stable/site
but the solution did not work.??? any suggestions ?

ilg-ul, 2017-06-05:
yes, use the latest java JDK and install the separate strong encryption package from oracle.

divyasujeeth, 2018-05-25:
Hi.. Need help. When I try to download package from Neon(URL : ) Im getting the below error. I have tried assigning proxy and also installed SSL Extender for further assistance with this. Unable to connect to repository.

[image]
[image]

ilg-ul, 2018-05-25:
if you have limited net access, you won't get too far.